OpenTabletDriver/OpenTabletDriver #2745: Support Request: Add Support for espresso Display

AA-espresso, 2023-05-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

Add support for espresso display on Linux Wayland and Xorg

### What have you tried?

I have tried building from source and make some changes to the configuration

### Diagnostic Information

[debug.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11527799/debug.txt)

AA-espresso, 2023-05-22:
Add Support for espresso display

jamesbt365, 2023-08-23:
If you mean what you said this is going to cause conflicts because the 2 espresso tablets are the same in identifiers. This will cause problems and will make differentiating between models cleanly a struggle, and would require manual intervention by the user in order to get the tablet to work.

https://discord.com/channels/615607687467761684/723399565780189234/1108184376451285012

Adding to backlog as this is low priority for us. I don't plan personally on creating a configuration for this tablet at this time because the issues above.