OpenTabletDriver/OpenTabletDriver #2756: add a new tablet Parblo Ninos N series

Zoliter, 2023-05-24:
### Description

add a new tablet Parblo Ninos N series, they are very popular in the osu community. many players or users of this tablet are waiting for it in the otd. please add it to the domain. Thanks for understanding :3

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-05-24:
Diagnostics are required for tablet support requests, either go to help > export diagnostic, save that file and send it here or reopen your issue and use the configurations label and follow the instructions on that and fill in the other important information.

jamesbt365, 2023-08-23:
Fixed by #2679 and #2734, nobody has asked or provided info on the N10 as of the current moment so closing this.

closed 2023-08-23 by jamesbt365