OpenTabletDriver/OpenTabletDriver #2757: OpenTabletDriver UI not opening no matter what (Linux Mint)

lost3o, 2023-05-24:
### Steps to Reproduce

1. Open OpenTabletDriver from the start menu or from command line

### Expected Behavior

UI opens (as it used to do, at least on my laptop)

### Observed Behavior

Nothing happens at all

### Tablet Device

Wacom CTL-672

### Diagnostic Information

Can't get, no UI

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-05-24:
If running the daemon says about missing libhostfxr then follow this.

https://learn.microsoft.com/en-us/dotnet/core/install/linux-package-mixup#solutions

lost3o, 2023-05-24:
Running the daemon says nothing either, didn't mention that

jamesbt365, 2023-08-23:
if dotnet --info lists a missing folder then please follow the link above.

closed 2023-08-23 by jamesbt365