OpenTabletDriver/OpenTabletDriver #2758: Add support for Wacom CTL-4400

DangerMage, 2023-05-24:
### Manufacturer

Wacom

### Product Name

CTL-440 (could be CTI? It's hard to read the black text on black tablet)

### Additional Information

I replaced my username in the json with `withheld`

I'm not sure if I correctly uninstalled wacom's drivers

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\withheld\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\withheld\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\withheld"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_17ef&pid_6099#6&9cd4084&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Primax",
      "ProductName": "Lenovo Traditional USB Keyboard",
      "SerialNumber": "",
      "FriendlyName": "Lenovo Traditional USB Keyboard",
      "VendorID": 6127,
      "ProductID": 24729,
      "InputReportLength": 9,
      "OutputReportLength": 2,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_045e&pid_0095&col03#6&258d57fa&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "",
      "ProductName": "",
      "SerialNumber": "",
      "FriendlyName": "",
      "VendorID": 1118,
      "ProductID": 149,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 2,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_0d8c&pid_016c&mi_02#7&1087da86&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Cyber Acoustics",
      "ProductName": "SHASTA",
      "SerialNumber": "REV1",
      "FriendlyName": "SHASTA",
      "VendorID": 3468,
      "ProductID": 364,
      "InputReportLength": 16,
      "OutputReportLength": 16,
[... 1603 lines cut ...]
    },
    {
      "Time": "2023-05-24T10:45:56.5895077-06:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-05-24T10:45:56.5895144-06:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-05-24:
The closest tablet to exist from that name is the CTL-4100, the CTL-4400 doesn't exist, and you don't have a tablet visibly plugged in from the diagnostic.

DangerMage, 2023-05-24:
[image]

[image]

It's 100% plugged in. I could 100% be reading it wrongly but it's hard to read on the plastic.

DangerMage, 2023-05-24:
[image]

[image]

Here's the driver details since I see it's being odd inside of device manager

DangerMage, 2023-05-24:
Ah thanks

closed 2023-05-24 by DangerMage