OpenTabletDriver/OpenTabletDriver #2766: Add support for HS611 w/ latest FW

adam-kulagowski, 2023-05-30:
### Manufacturer

Huion

### Product Name

HS611

### Device Width

258.4mm

### Device Height

161.5mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Latest FW on HS611 changes USB product ID of upgraded tablet. This makes the device unusable in Linux unless one want to use closed source Huion linux drivers.  New USB ID is:

`256c:006f HUION Huion Tablet_HS611`

Attached new configuration, placed in `/usr/share/OpenTabletDriver/Configurations/HS611-fw.json` makes the tablet fully visible and operational again for OpenTabletDriver:

[HS611-new-fw.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11597282/HS611-new-fw.txt)

I've created this configuration based on original HS611 config from github. The only things I've changed are:
`ProductID` and `DeviceStrings/201`

I'm attaching also device strings dump:
[hs611-dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11597315/hs611-dump.txt)

### Diagnostic Information

[hs611-diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11597321/hs611-diag.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

adam-kulagowski, 2023-05-30:
I'm attaching updated configuration with tilt working. 

[HS611-fw-tilt.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11598827/HS611-fw-tilt.txt)

X9VoiD, 2023-08-16:
Resolved by #2801

closed 2023-08-16 by X9VoiD