OpenTabletDriver/OpenTabletDriver #2768: Support Request: No tablets are detected, Gaomon S620, NixOS 23.05

nightingale-m, 2023-06-02:
### What do you need help with?

Tablet isn't detected. OTD worked with the same tablet on Archlinux.

OS - NixOS 23.05
OTD v0.6.0.4
Tablet - Gaomon S620

### What have you tried?

I generated and added udev rules:
```nix
{
  services.udev.extraRules = ''
    # Gaomon S620
    SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", MODE="0666"
    SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", MODE="0666"
    SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", MODE="0666"
    SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", MODE="0666"
    SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", ENV{LIBINPUT_IGNORE_DEVICE}="1"
    SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", ENV{LIBINPUT_IGNORE_DEVICE}="1"
  '';
}
```

### Diagnostic Information

[otd-diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11635914/otd-diagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

closed 2023-06-02 by jamesbt365

nightingale-m, 2023-06-02:
Copying configuration file solved the problem. Thank you!

glow12121, 2023-06-14:
I've added `Gaomon.json` to `~/.config/OpenTabletDriver/Configurations/` and also `~/.config/OpenTabletDriver/Configurations/Gaomon/` but to no avail.

I use Fedora 38 with Gnome. It is a Wayland system.

The console output of OTD 0.6.0.4:

``
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
``

jamesbt365, 2023-06-14:
`/usr/share/OpenTabletDriver/Configurations`. You will also need new udev rules.

glow12121, 2023-06-14:
I downloaded the source code and run:

```
$ sudo sh generate-rules.sh 
Generating udev rules...
The command could not be loaded, possibly because:
  * You intended to execute a .NET application:
      The application 'run' does not exist.
  * You intended to execute a .NET SDK command:
      No .NET SDKs were found.
```

However, the following dependencies were added during the driver's RPM installation:

```
Package                        Arch    Version                    Repository        Size
Installing:
 opentabletdriver              x86_64  0.6.0.4-1                  @commandline  25.2 MiB
Installing dependencies:
 dotnet-host                   x86_64  7.0.5-1.fc38               updates      296.2 KiB
 dotnet-hostfxr-6.0            x86_64  6.0.16-1.fc38              updates      381.7 KiB
 dotnet-runtime-6.0            x86_64  6.0.16-1.fc38              updates       65.0 MiB
 libXau-devel                  x86_64  1.0.11-2.fc38              fedora         6.4 KiB
 libXext-devel                 x86_64  1.3.5-2.fc38               fedora        98.8 KiB
 libXrender-devel              x86_64  0.9.11-2.fc38              fedora        50.1 KiB
 libevdev-devel                x86_64  1.13.1-1.fc38              updates       95.6 KiB
 libxcb-devel                  x86_64  1.13.1-11.fc38             fedora         2.6 MiB
 xorg-x11-proto-devel          noarch  2022.2-3.fc38              fedora         1.7 MiB
Installing weak dependencies:
 libX11-devel                  x86_64  1.8.4-1.fc38               fedora         1.0 MiB
 libXrandr-devel               x86_64  1.5.2-10.fc38              fedora        21.9 KiB
```

jamesbt365, 2023-06-14:
you need the dotnet 7 sdk to run the script.

glow12121, 2023-06-14:
I successfully ran:

`$ sudo dnf5 install dotnet-sdk-7.0`

`%sh generate-rules.sh`

`$ sudo mv ~/Downloads/OpenTabletDriver-master/bin/99-opentabletdriver.rules /etc/udev/rules.d/`

`$ sudo udevadm control --reload-rules`

I rebooted too - but all this didn't help. Same output from OTD.

Also, for some reason every time OTD starts it displays the welcome screen.  And the console output is: `Presets have been refreshed. Loaded 0 presets.`

glow12121, 2023-06-14:
Diagnostics output: https://gitlab.com/-/snippets/2556199