OpenTabletDriver/OpenTabletDriver #2769: Wacom CTL-6100WL Auxiliary buttons don't work

Arisa-Snowbell, 2023-06-02:
### Steps to Reproduce

Set bindings to the keys and try them

### Expected Behavior

To execute the key combination I set on the buttons

### Observed Behavior

Nothing happens

### Tablet Device

Wacom CTL-6100WL

### Diagnostic Information

I won't be providing this whole file, the file contains sensitive info like my environment variables and to censor my self the entire file is impossible because of the console log the file is enormous. I am using the wired connection for this, haven't tried wireless, but I don't require that, and I see it's missing wireless support, but there is nothing about buttons missing support. The two pen buttons work but the four buttons on the tablet itself not, should I provide raw tablet data for each button press?

```json

{
  "App Version": "OpenTabletDriver v0.6.0.4+19803e0",
  "Build Date": "5/30/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.3.5.1",
    "VersionString": "Unix 6.3.5.1"
  },
  "Environment Variables": {
    "USER": "why need this",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "NOPE"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:05:00.3/usb1/1-2/1-2:1.0/0003:056A:0378.0005/hidraw/hidraw2",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos BT M",
      "SerialNumber": "why need this?",
      "FriendlyName": "Intuos BT M",
      "VendorID": 1386,
      "ProductID": 888,
      "InputReportLength": 192,
      "OutputReportLength": 0,
      "FeatureReportLength": 1281,
      "CanOpen": true
    }
  ],

```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-08-23:
Do the "aux buttons" go to true when you press them?

jamesbt365, 2023-11-10:
Closing as missing details and because its probably solved in the latest version.

closed 2023-11-10 by jamesbt365