OpenTabletDriver/OpenTabletDriver #2777: can't build udev rules

tokisuno, 2023-06-08:
### Steps to Reproduce

downloaded otd from the linux guide. restarted my computer. ran ./generate-rules.sh

### Expected Behavior

i have rules

### Observed Behavior

no rules
[image]

### Tablet Device

Wacom CTL-470

### Diagnostic Information

```[Detect:Info]	Searching for tablets...
[Detect:Info]	Found tablet 'Wacom CTL-470'
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 1740>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Wacom CTL-470:Info]	Output mode: Absolute Mode
[Wacom CTL-470:Info]	Display area: [1920x1080@<960, 540>:0°],
[Wacom CTL-470:Info]	Tablet area: [147.2x92@<73.6, 46>:0°],
[Wacom CTL-470:Info]	Clipping: Enabled
[Wacom CTL-470:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-470:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[... 89 lines cut ...]
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d)
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxOutputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_OutputReportLength>b__9_0(HidDevice d)
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxFeatureReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_FeatureReportLength>b__11_0(HidDevice d)
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value)
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-06-08:
As I said to you previously, you do not have the dotnet 7 sdk, which is required as the error message says.

tokisuno, 2023-06-08:
as you've said to me previously? this is my first time here. also does that mean the dotnet 6 is used for building opentabletdriver while dotnet 7 is used for generating rules if needed?

this should probs be on the linux faq https://opentabletdriver.net/Wiki/FAQ/Linux

tokisuno, 2023-06-08:
as it turns out i have dementia and forgot i posted an issue request already. that's my bad LMOA