OpenTabletDriver/OpenTabletDriver #2779: Opening OTD with keyboard causes error, but not with mouse.

aL1NaZ4R, 2023-06-09:
### Steps to Reproduce

Watch this video. I'm opening OTD with keyboard buttons and I get error on launch.
Then, I opened it with the mouse and OTD is fine and good.
https://github.com/OpenTabletDriver/OpenTabletDriver/assets/89677246/3a61a58f-87a3-434f-ae16-739364e68fae

### Expected Behavior

Shouldn't it run correctly with keyboard?

### Observed Behavior

Error (see in video)

### Tablet Device

XP-Pen G640S (Not V2)

### Diagnostic Information

[t.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11706455/t.json.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-06-09:
Does this happen when you do it with the actual binaries and not the shortcut?

aL1NaZ4R, 2023-06-09:
Yeah.

gonX, 2026-10-02:
Does this still happen on v0.6.7 or a development version? We're close to releasing v0.6.8 so a development version might be better to test

rvnexe, 2026-10-03:
I've switched to native drivers, but will try OTD's asap🙏

Mrcubix, 2026-10-03:
Unable to reproduce in 0.6.7 on my main machine, will see if i can reproduce on a lower-end machine later.
Doubt i will be able to reproduce it, probably an issue specific to 0.6.0.4 .

rvnexe, 2026-10-05:
It's been fixed. I can't reproduce on 0.6.7

rvnexe, 2026-10-05:
oooh no

rvnexe, 2026-10-05:
I tested with the binary package itself; and it didn't reproduce.

But then made a shortcut on my desktop (as I been doing back then in 2023) and still got error on launch.

[image]

Kuuuube, 2026-10-05:
How did you create the shortcut? If you right click the binary, click create shortcut, and try running that it should work. Usually this issue happens because the working directory is not correctly set in the shortcut.

rvnexe, 2026-10-05:
Ah got it.

closed 2026-10-05 by rvnexe