OpenTabletDriver/OpenTabletDriver #2784: cant find OpenTabletDriver.Plugin error when opening otd UI, cant use tablet

ana-rchy, 2023-06-11:
### Steps to Reproduce

1. install opentabletdriver-git from AUR
2. run `systemctl --user enable --now opentabletdriver.service`
3. open otd UI

### Expected Behavior

tablet should be usable, UI should open normally

### Observed Behavior

tablet doesnt work, UI errors out with log given below

### Tablet Device

wacom ctl-472

### Diagnostic Information

[crash.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11714834/crash.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

hwsmm, 2023-06-11:
Use `opentabletdriver` instead of `-git` for now, git version doesn't work well at the moment

jamesbt365, 2023-06-11:
Remove `~/.config/OpenTabletDriver` and use the `opentabletdriver` package, master is currently not stable and cases where trying to use outdated plugins will cause it to crash (master doesn't have any plugins for it either)

ana-rchy, 2023-06-11:
> Remove `~/.config/OpenTabletDriver` and use the `opentabletdriver` package, master is currently not stable and cases where trying to use outdated plugins will cause it to crash (master doesn't have any plugins for it either)

for future reference to other people, this seems to work

jamesbt365, 2023-06-11:
duplicate of #2407

closed 2025-12-05 by gonX