OpenTabletDriver/OpenTabletDriver #2790: Select mapping to a single monitor

mikesTech, 2023-06-15:
### Description

It would be nice to select a mapping to a single monitor, so people who regularly disconnect displays don't have to constantly resize using the area editor whenever a monitor disconnects (as the areas mess up 
[image]
)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Phyan, 2023-08-18:
Yes, I have the same problem. Use of multiple monitors in combination with OTD causes a lot of inconvenience because of it.

KelvinNovais, 2024-06-30:
I have the same problem:
If it starts with 2 monitors, and I disconnect the second monitor, OTD sets the display size to the half of the builtin monitor;
If it starts only with the builtin monitor and I connect the second one, OTD sets the display size to the monitors together;
I can't list all cases, but when I switch or join monitors, OTD can't handle well with that.

An anchor would be nice, specially if you can priority which monitor use; so when a monitor is disconnected, OTD automatically switches to another.

gonX, 2026-09-25:
Narrowly related issue: #1143

Moving this out of the Avalonia milestone as this is interesting to fix in mainline OTD