OpenTabletDriver/OpenTabletDriver #2791: OTD cannot detect Huion Q620M, "No tablets were detected", Arch Linux

geothecode, 2023-06-17:
### Steps to Reproduce

1. Connect tablet's usb connector (i.e. bluetooth stick)
2. Check if it's indeed connected using `lsusb` or `dmesg`
3. It is connected => launch OpenTablet using `gtk-launch OpenTabletDriver`
4. Wait for it to load and do not close the terminal window/tile

### Expected Behavior

I expect my tablet to be detected not only by the kernel but by the OpenTabletDriver too

### Observed Behavior

And so instead I see that my tablet is detected only by the system (i.e. it shows up in the lsusb and dmesg outputs: e.g. `[181.336731] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[181.336734] usb 2-2: Product: Huion tablet_Q620M
[181.336737] usb 2-2: Manufacturer: HUION`), while OpenTabletDriver (OTD) does not detect it, although it searches for it (as it is shown in the logs). And one more detail: OTD gives me these errors and warnings: `(OpenTabletDriver.UX.Gtk:14629): libnotify-WARNING **: 18:53:00.390: Failed to connect to proxy` and this one when I try forcing it to find my tablet by Ctrl-D (detect tablet) `No tablets were detected`. And... It constantly shows this error: `[ArgumentOutOfRangeException:Error]	Value range is [0, 15]. (Parameter 'value')
   at HidSharp.Reports.Units.Unit.DecodeExponent(UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseMainData(MainItemTag tag, UInt32 value)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItem(EncodedItem item)
   at HidSharp.Reports.ReportDescriptor.ParseEncodedItems(IEnumerable1 items)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func 2 predicate, TValue& value)`

### Tablet Device

Huion Q620M (usb name: HUION Huion tablet_Q620M)

### Diagnostic Information

[opentabletdiagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11779408/opentabletdiagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-06-17:
You have the hid_uclogic module loaded, see [this](https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception) wiki entry to figure out how to solve this issue.

Also duplicates of #2775, #2647, #2355, #949, #870, #760, #517, #452, #408, #355 and more.

closed 2023-06-17 by jamesbt365

geothecode, 2023-06-17:
goodness, I'm so sorry, read it twice and still hadn't even noticed... thanks for your support!! maybe it would've been better to highlight it somehow for people like me to notice in a first place :)