OpenTabletDriver/OpenTabletDriver #2798: Add support for Huion Inspiroy RTS-300 tablet.

ItzMehNub, 2023-06-22:
### Manufacturer

Huion

### Product Name

Inspiroy RTS-300

### Device Width

200 x 163 mm (160 x 100 mm for workspace)

### Device Height

8.35 mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Was mentioned in #2063 about sharing a name with the Huion H642, which it did. but support was still lacking. I don't really know if this counts as a duplicate or not.

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.1.1",
  "Build Date": "6/21/2023",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\ASUS\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\ASUS\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\ASUS"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_03eb&pid_8417&mi_01#8&1fda449a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Atmel",
      "ProductName": "Atmel maXTouch Control",
      "SerialNumber": "",
      "FriendlyName": "Atmel maXTouch Control",
      "VendorID": 1003,
      "ProductID": 33815,
      "InputReportLength": 64,
      "OutputReportLength": 64,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_30fa&pid_1701&mi_01&col02#7&140762ac&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "INSTANT",
      "ProductName": "USB GAMING MOUSE ",
      "SerialNumber": "",
      "FriendlyName": "USB GAMING MOUSE ",
      "VendorID": 12538,
      "ProductID": 5889,
      "InputReportLength": 3,
[... 2120 lines cut ...]
    },
    {
      "Time": "2023-06-22T10:04:43.6268165+08:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-06-22T10:04:43.6268335+08:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-08-23:
tablet > device string reader
9580
109

dump all (leave the bottom box blank)

save that file and sent it here.

ItzMehNub, 2023-11-28:
alright. here it is
[Huion RTS-300.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13483865/Huion.RTS-300.txt)

also this happened out of nowhere, it was unfocused and minimized, it then suddenly did this and sent out like 50 error popups. I have no idea what caused it nor how to reproduce, and it crashed as I'm about to copy the logs.
[image]

vzade-dev, 2024-05-22:
Randomly decided to try this again, now it actually works under Huion H642 with no problems as of v0.6.4.0. Anything else I might need to try out?

closed 2024-12-29 by jamesbt365