OpenTabletDriver/OpenTabletDriver #2799: OpenTabletDriver Updater Error

HashDub, 2023-06-22:
### Steps to Reproduce

Tried updating my current version (v0.6.1.0) to latest (v0.6.1.1) using the Updater. Doing so resulted in an application error.

### Expected Behavior

I expect for the program to update itself without any issues.

### Observed Behavior

Application error popup window with code.

### Tablet Device

Wacom CTH-480

### Diagnostic Information

``StreamJsonRpc.RemoteInvocationException: The file exists.
   at StreamJsonRpc.JsonRpc.InvokeCoreAsync[TResult](RequestId id, String targetName, IReadOnlyList`1 arguments, IReadOnlyList`1 positionalArgumentDeclaredTypes, IReadOnlyDictionary`2 namedArgumentDeclaredTypes, CancellationToken cancellationToken, Boolean isParameterObject)
   at OpenTabletDriver.UX.Windows.Updater.UpdaterWindow.<>c__DisplayClass4_0.<<Update>b__0>d.MoveNext() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Windows\Updater\UpdaterWindow.cs:line 80
--- End of stack trace from previous location ---
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_0(Object state)
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)
RPC server exception:
System.IO.IOException: The file exists.
      at System.IO.FileSystem.MoveFile(String sourceFullPath, String destFullPath, Boolean overwrite)
      at System.IO.File.Move(String sourceFileName, String destFileName, Boolean overwrite)
      at OpenTabletDriver.Desktop.Updater.Updater.InstallUpdate(Update update) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Updater\Updater.cs:line 91
      at OpenTabletDriver.Desktop.Updater.Updater.Install(Update update) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Updater\Updater.cs:line 67
      at OpenTabletDriver.Daemon.DriverDaemon.InstallUpdate() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Daemon\DriverDaemon.cs:line 479``

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

HashDub, 2023-06-22:
Also noticed that in the application console, it said that it only supported a frequency of 125, 250, 500, or 1000 ms in the Devocub anti-chatter settings i was using.

X9VoiD, 2023-06-22:
You have a system that doesn't allow moving files while they have open handles. Download the update manually for now.

closed 2023-06-22 by X9VoiD