OpenTabletDriver/OpenTabletDriver #2800: Crash on startup

solonovamax, 2023-06-22:
### Steps to Reproduce

Launch open tablet driver

### Expected Behavior

Doesn't crash

### Observed Behavior

Crashes

### Tablet Device

XP-Pen Star G640 V2

### Diagnostic Information

```
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S'
[Evdev:Debug]	Successfully initialized virtual pressure sensitive tablet. (code NONE)
[XP-Pen Star G640 V2:Info]	Output mode: Artist Mode
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at OpenTabletDriver.Daemon.DriverDaemon.SetAbsoluteModeSettings(InputDeviceTree dev, AbsoluteOutputMode absoluteMode, AbsoluteModeSettings settings)
   at OpenTabletDriver.Daemon.DriverDaemon.SetSettings(Settings settings)
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
/usr/bin/otd: line 13: 27563 Aborted                 (core dumped) dotnet ./OpenTabletDriver.Daemon.dll
```

perhaps this has something to do with reading the settings?
so, here's my settings:
```json
{
  "Revision": "0.7.0.0",
  "Profiles": [
    {
      "Tablet": "XP-Pen Star G640 V2",
      "OutputMode": {
        "Path": "OpenTabletDriver.Desktop.Output.LinuxArtistMode",
        "Settings": [
          {
            "Property": "Input",
            "Value": {
              "Rotation": 0.0,
              "Width": 160.0,
              "Height": 100.0,
              "XPosition": 80.0,
              "YPosition": 50.0
            }
          },
          {
            "Property": "Output",
            "Value": {
              "Width": 1920.0,
              "Height": 1080.0,
              "XPosition": 2880.0,
              "YPosition": 540.0
            }
          },
          {
            "Property": "LockAspectRatio",
            "Value": false
          },
[... 279 lines cut ...]
              {
                "Property": "Button",
                "Value": "Middle"
              }
            ],
            "Enable": true
          }
        ],
        "AuxButtons": [],
        "MouseButtons": [],
        "MouseScrollUp": null,
        "MouseScrollDown": null
      }
    }
  ],
  "Tools": []
}
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

solonovamax, 2023-06-22:
I know the issue isn't very informative, but I don't have a lot to work on

it just always crashes when I run the `otd` command. (also the systemd script crashes as well, but that just runs `odt`, I assume)

X9VoiD, 2023-06-22:
Reset settings, you're using 0.7.x settings on a 0.6.x daemon. They have incompatible settings format.

closed 2023-06-22 by X9VoiD

solonovamax, 2023-06-22:
will do, ty

solonovamax, 2023-06-22:
not sure if you're the maintainer of it or not, but the aur package for `opentabletdriver` seems to be packaging 0.6.0.4

X9VoiD, 2023-06-22:
Already known, just wait till maintainers update the package.