OpenTabletDriver/OpenTabletDriver #2806: When I draw a line with Wacom CTL-490, it always squiggle weirldy at it's end.

zor1an, 2023-06-24:
### Steps to Reproduce

1. Open OpenTabletDriver
2. Take Wacom CTL-490
3. Try to draw a quick bent line

### Expected Behavior

Well, I expect to draw a line without any kind of squiggles at it's end. It works just fine in Wacom drivers:
[image]

### Observed Behavior

But in OpenTabletDriver, they do have some kind of squiggle at it's end. Here is an example:
[image]

### Tablet Device

CTL-490

### Diagnostic Information

[debug.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11854157/debug.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Kuuuube, 2023-06-24:
Currently, you can use Slimy Scylla plugin as a workaround. The following settings will add no smoothing but will stop those bad lines being drawn.

[image]

If you are using windows ink plugin you can set `Remove Tail Pressure Reports` to `1` instead of `2`.

This issue will be kept open for tracking.

zor1an, 2023-06-25:
Thanks for help! Works fine now.

zor1an, 2023-06-25:
Sorry, I'm stupid lol