OpenTabletDriver/OpenTabletDriver #2824: Support Request: support for XP-penartist pro 16 Gen 1

Dmeerev, 2023-07-02:
### What do you need help with?

Requesting support for XP-penartist pro 16 Gen 1 

in zadig i have 3 devices:
tablet monitor (interface 0) usbid 5543 004D 00
tablet monitor (interface 1) usbid 5543 004D 01
tablet monitor (interface 2) usbid 5543 004D 01

img:
[https://cdn.discordapp.com/attachments/615611007951306863/1124852770705514506/154864430967bfa1919ab21457.png](url)

### What have you tried?

I tried installing latest OTD and it dont detect my tablet, in ODT database usbid matches with xp pen artist 16 pro gen 2, but gen 1 is not working in any way.

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.2.0",
  "Build Date": "6/25/2023",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19045.0",
    "VersionString": "Microsoft Windows NT 10.0.19045.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\dmeerev\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\dmeerev\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\dmeerev"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_004d&mi_02&col01#7&9a93b81&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "UGEE",
      "ProductName": "TABLET MONITOR",
      "SerialNumber": "",
      "FriendlyName": "TABLET MONITOR",
      "VendorID": 21827,
      "ProductID": 77,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_04f3&pid_1213&mi_01&col01#8&3436844f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 1267,
      "ProductID": 4627,
      "InputReportLength": 6,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_5543&pid_004d&mi_01&col02#7&1a2e0241&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "UGEE",
      "ProductName": "TABLET MONITOR",
      "SerialNumber": "",
      "FriendlyName": "TABLET MONITOR",
      "VendorID": 21827,
      "ProductID": 77,
      "InputReportLength": 0,
[... 2077 lines cut ...]
    {
      "Time": "2023-07-02T03:47:05.3017709+03:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-07-02T03:47:05.3017763+03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Dmeerev, 2023-07-04:
So, any way to make tablet work with this driver?

jamesbt365, 2023-08-23:
https://opentabletdriver.net/Wiki/Install/Windows#winusb

Follow these instructions and install winusb on **only** interface 0, send another diagnostic once completed.

m10d, 2023-09-29:
Is this issue for XPPen artist **pro** gen **1** or gen **2**?

title & some content references "gen1" but also "gen2" / "2nd"

Just checking before I open a new request for support, thanks.

```
[61511.863379] usb 5-4.1: New USB device found, idVendor=28bd, idProduct=095b, bcdDevice= 0.00
[61511.863385] usb 5-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[61511.863387] usb 5-4.1: Product: Artist Pro 16 (Gen2)
[61511.863389] usb 5-4.1: Manufacturer: UGTABLET
[61512.010974] input: UGTABLET Artist Pro 16 (Gen2) Mouse as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-4/5-4.1/5-4.1:1.0/0003:28BD:095B.0014/input/input49
[61512.011189] input: UGTABLET Artist Pro 16 (Gen2) Keyboard as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-4/5-4.1/5-4.1:1.0/0003:28BD:095B.0014/input/input50
[61512.071421] hid-generic 0003:28BD:095B.0014: input,hidraw5: USB HID v1.00 Mouse [UGTABLET Artist Pro 16 (Gen2)] on usb-0000:0b:00.3-4.1/input0
[61512.078589] input: UGTABLET Artist Pro 16 (Gen2) as /devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-4/5-4.1/5-4.1:1.1/0003:28BD:095B.0015/input/input51
[61512.078689] hid-generic 0003:28BD:095B.0015: input,hidraw6: USB HID v1.00 Device [UGTABLET Artist Pro 16 (Gen2)] on usb-0000:0b:00.3-4.1/input1
[61512.086561] hid-generic 0003:28BD:095B.0016: hiddev2,hidraw7: USB HID v1.00 Device [UGTABLET Artist Pro 16 (Gen2)] on usb-0000:0b:00.3-4.1/input2
```

jamesbt365, 2023-09-29:
These are different tablets, open an independent configuration request as "artist Pro 16 Gen 2".

m10d, 2023-09-30:
> These are different tablets, open an independent configuration request as "artist Pro 16 Gen 2".

done, thanks

Dmeerev, 2024-02-09:
> https://opentabletdriver.net/Wiki/Install/Windows#winusb
> 
> Follow these instructions and install winusb on **only** interface 0, send another diagnostic once completed.

[5543.004d.00.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14220549/5543.004d.00.json)

folliehiyuki, 2024-03-03:
Hi @Dmeerev ! Is this request for https://www.xp-pen.com/product/artist-16-pro.html?

The support for [Artist Pro 16 Gen1](https://www.xp-pen.com/product/artist-pro-16.html) was added in https://github.com/OpenTabletDriver/OpenTabletDriver/pull/2332 (but it was incorrectly named **Artist 16 Pro**).