OpenTabletDriver/OpenTabletDriver #2827: Unable to install 0.6.2 rpm in openSUSE

FvD, 2023-07-02:
### Steps to Reproduce

Dowload the latest rpm v0.6.2.0 on opensuse tumbleweedand trying to install with 

`sudo zypper install OpenTabletDriver.rpm`

### Expected Behavior

Given the announcement in the changelog that opensuse tumbleweed is not supported I expected all dependencies to be resolved by zypper.

### Observed Behavior

`sudo zypper install OpenTabletDriver.rpm`

gives a warning that the dependency `dotnet-runtime-6.0` cannot be met.

I have not found `dotnet-runtime-6.0` in the zypper repos and tried the [installation by using an installscript](https://learn.microsoft.com/en-us/dotnet/core/install/linux-scripted-manual)  and set the environment variables as outlined on that page, before attempting to install again.

```
export DOTNET_ROOT=$HOME/.dotnet
export PATH=$PATH:$DOTNET_ROOT:$DOTNET_ROOT/tools
```

But this does not resolve the dependency. Ignoring the warning and proceeding with the installation does not seem to lead to a working server (`systemctl start | status opentabletdriver.service` does not work).

### Tablet Device

HUION WH1409v2

### Diagnostic Information

non available

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

X9VoiD, 2023-07-03:
https://learn.microsoft.com/en-us/dotnet/core/install/linux-opensuse

closed 2023-07-03 by X9VoiD

maeries, 2025-03-26:
Following that guide I get the issue that dotnet-runtime depends on libopenssl1_0_0 which I don't know how to install

jamesbt365, 2025-03-26:
We don't support this version anymore.