OpenTabletDriver/OpenTabletDriver #2829: Support Request: support for Parblo A640

ClaireRoush, 2023-07-03:
### What do you need help with?

Requesting support for Parblo A640, because list of supported device's only have v2 version. 

### What have you tried?

I tried installing latest OTD and it don't detect my tablet. 

### Diagnostic Information

[meow.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11940039/meow.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-08-23:
I can't find very much information on this tablet online, so here is a configuration that will hopefully init the tablet successfully without any tweaks.

Download [this](https://cdn.discordapp.com/attachments/628651971267788800/1144013661191553235/A640.json) file, create a folder called `Configurations` next to the executables of OpenTabletDriver (`OpenTabletDriver.Daemon.exe`, `OpenTabletDriver.UX.Wpf.exe`, etc, etc) and place this file into this folder, replug your tablet or restart OpenTabletDriver once done.

Go into OpenTabletDriver, navigate to the top where it says tablets, click that and open the `Tablet Debugger` and try and get the highest max x and y coordinate that you can, please tell me the results of that here.

I'm aware that the pen buttons will not be bindable at the current time, but they should appear in the tablet debugger when you press the buttons. The parser should also be correct.

ClaireRoush, 2023-09-07:
hm... i think its 75 and 45. 
[image]

ClaireRoush, 2023-09-07:
Tablet debugger doesn't open when i click on it.

jamesbt365, 2023-09-07:
Then you need to follow the instructions you see when you open the driver, and not ignore them.

ClaireRoush, 2023-09-09:
in tablet debugger are nothing :<

3ab3z2, 2023-12-23:
are there any updates? I have this tablet too, but I'm on linux (the official driver works just fine on windows, there isn't any for linux)

jamesbt365, 2024-07-30:
Fixed by #3157.

closed 2024-07-30 by jamesbt365