OpenTabletDriver/OpenTabletDriver #2838: Plugins refuse to install.

ajd07, 2023-07-18:
### Steps to Reproduce

I have reinstalled opentabletdriver before already and this has only happened on the **latest** build for me. The information provided is similar to another post that I saw but is not the same error I was given.

Open the UI
Go to plugin manager
Attempt to install a plugin

### Expected Behavior

It should have installed the plugin just fine.

### Observed Behavior

It throws out an error, of course.

### Tablet Device

Huion H420

### Diagnostic Information

`[System.BadImageFormatException:Error]	Could not load file or assembly 'ICSharpCode.SharpZipLib, Version=1.3.3.11, Culture=neutral, PublicKeyToken=1b03e6acf1164f73'. The module was expected to contain an assembly manifest.
   at OpenTabletDriver.Desktop.Compression.CompressionTools.Decompress(Stream stream, String outputDir, String format)
   at OpenTabletDriver.Desktop.Reflection.Metadata.PluginMetadata.DownloadAsync(String outputDirectory) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Reflection\Metadata\PluginMetadata.cs:line 104
   at OpenTabletDriver.Desktop.Reflection.DesktopPluginManager.DownloadPlugin(PluginMetadata metadata) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.Desktop\Reflection\DesktopPluginManager.cs:line 185
`

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

ajd07, 2023-07-18:
if you ask, this is on windows 11.

gonX, 2023-10-02:
Please try again on the latest version of OpenTabletDriver (currently 0.6.3.0), also you should post the full diagnostics as the template suggests - this is why it got tagged `missing-details`.

closed 2023-10-02 by gonX