OpenTabletDriver/OpenTabletDriver #2856: Support Request: OPT seems to detect my tablet but the settings don't apply( on a wacom ctl-472)

PovoloRN, 2023-07-28:
### What do you need help with?

When i try to use my tablet  i usually open the driver and eveything is fine. But in the last week whenever i open the drivers, it looks like they have detected the tablet, but then the settings don't apply to it (like the area that i choose ecc...).
It's like using a new tablet without drivers.
I use a wacom ctl-472.This problem started after i had an issue with my graphic card( some drivers were not updated so the graphic wasn't working). and after  i fixed this issue started.

### What have you tried?

i tried to restart my pc and it didn't work.
I tried to update my usb drivers and nothing changed. 
I  searched  on the internet if someone with a ctl-472 had  the same problem but i didn't found anything helpfull.
i tried to swith usb port but it still wasn't working.

### Diagnostic Information

sad

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

PovoloRN, 2023-07-28:
{
  "App Version": "OpenTabletDriver v0.6.2.0",
  "Build Date": "6/25/2023",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22621.0",
    "VersionString": "Microsoft Windows NT 10.0.22621.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\PC\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\PC\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\PC"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_056a&pid_037a&col01#7&1caa568&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "CTL-472",
      "SerialNumber": "1GA00L1079763",
      "FriendlyName": "CTL-472",
      "VendorID": 1386,
      "ProductID": 890,
      "InputReportLength": 7,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_03f0&pid_0c8e&mi_01&col05#8&856e90c&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "HP, Inc",
      "ProductName": "HyperX Alloy Origins 60",
      "SerialNumber": "",
      "FriendlyName": "HyperX Alloy Origins 60",
      "VendorID": 1008,
      "ProductID": 3214,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#elan0636&col01#4&2b48d8a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Microsoft",
      "ProductName": "HIDI2C Device",
      "SerialNumber": "9999",
      "FriendlyName": "HIDI2C Device",
      "VendorID": 1267,
      "ProductID": 1024,
      "InputReportLength": 9,
      "OutputReportLength": 0,
      "FeatureReportLength": 5,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_056a&pid_037a&col02#7&1caa568&1&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "CTL-472",
      "SerialNumber": "1GA00L1079763",
      "FriendlyName": "CTL-472",
      "VendorID": 1386,
      "ProductID": 890,
      "InputReportLength": 11,
      "OutputReportLength": 161,
      "FeatureReportLength": 1024,
      "CanOpen": true
    },
    {
[... 2273 lines cut ...]
      "Time": "2023-07-27T15:15:33.8250099+02:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetSerialNumber()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Devices\\HidSharpBackend\\HidSharpEndpoint.cs:line 25\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Extensions.cs:line 12",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2023-07-27T15:15:33.8250192+02:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetSerialNumber()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Devices\\HidSharpBackend\\HidSharpEndpoint.cs:line 25\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Extensions.cs:line 12",
      "Level": 3,
      "Notification": false
    },
    {
      "Time": "2023-07-27T15:15:33.8250285+02:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetSerialNumber()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_SerialNumber>b__19_0(HidDevice d) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Devices\\HidSharpBackend\\HidSharpEndpoint.cs:line 25\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Extensions.cs:line 12",
      "Level": 3,
      "Notification": false
    }
  ]
}

jamesbt365, 2023-07-28:
Duplicate of many issues.

Solution: https://github.com/X9VoiD/TabletDriverCleanup/releases/latest (this is listed on our frequently asked questions page)

closed 2023-07-28 by jamesbt365