OpenTabletDriver/OpenTabletDriver #2858: SUP/BUG: Tablet not detecting, app window will not open // Ugee S640

xendyex, 2023-07-30:
### Steps to Reproduce

All I know is I install Ubuntu 23.04, then OTD. I also have Ugee's own drivers, for my tablet, but still nothing will detect it.
So I need to open the app.
So open the app.

### Expected Behavior

It should have opened the window.

### Observed Behavior

It didn't open. It loaded in the top taskbar, then crashed, if that makes any sense.

### Tablet Device

Ugee S640/S640W

### Diagnostic Information

N/A

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-07-30:
Diagnostic information is applicable and I will not be able to properly help you until these are provided, file > export diagnostics, save that file and send it here, as the instructions told you to do.

xendyex, 2023-07-30:
> Diagnostic information is applicable and I will not be able to properly help you until these are provided, file > export diagnostics, save that file and send it here, as the instructions told you to do.

I can't open the app to get these, what?

jamesbt365, 2023-07-30:
If the app doesn't open, check the daemon, if its a dotnet installation issue follow here. https://learn.microsoft.com/en-us/dotnet/core/install/linux-package-mixup#solutions

xendyex, 2023-07-30:
Also, my tablet SHOULD be supported, so that's off the list.
[image]

xendyex, 2023-07-30:
> If the app doesn't open, check the daemon, if its a dotnet installation issue follow here. https://learn.microsoft.com/en-us/dotnet/core/install/linux-package-mixup#solutions

Ok, thanks, will try.

xendyex, 2023-07-30:
> If the app doesn't open, check the daemon, if its a dotnet installation issue follow here. https://learn.microsoft.com/en-us/dotnet/core/install/linux-package-mixup#solutions

N: Download is performed unsandboxed as root as file '/home/[my name]/Desktop/files/pen drivers/OpenTabletDriver.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
maybe this is the problem? it lets me continue, its when running
sudo apt install ./OpenTabletDriver.deb

jamesbt365, 2023-07-30:
Its a notice, not an error, just continue.

xendyex, 2023-07-30:
> Its a notice, not an error, just continue.

Ok. This time I got even worse - the app doesn't even appear, following

xendyex, 2023-07-30:
oh uhh
what
did i do

xendyex, 2023-07-30:
> Its a notice, not an error, just continue.

Even worse, app doesn't even appear now. Imma restart, one second...

xendyex, 2023-07-30:
> Its a notice, not an error, just continue.

Same thing - maybe I should check the daemon?

jamesbt365, 2023-07-30:
Check the message when you start the daemon.

xendyex, 2023-07-30:
> Check the message when you start the daemon.

I'm not the best at computers, sorry. I tried to see how to check it, but I got confused, and I'm more confused with this. Sorry, I know I'm annoying, but it's just me and I  can't not be me.

jamesbt365, 2023-07-30:
`otd-daemon` in a terminal.

xendyex, 2023-07-30:
huh...
[image]

i'll install .net then..? i thought i had

jamesbt365, 2023-07-30:
`dotnet --info`.

xendyex, 2023-07-30:
> `dotnet --info`.

$ dotnet

Usage: dotnet [options]
Usage: dotnet [path-to-application]

Options:
  -h|--help         Display help.
  --info            Display .NET information.
  --list-sdks       Display the installed SDKs.
  --list-runtimes   Display the installed runtimes.

path-to-application:
  The path to an application .dll file to execute.
me@me:~/Desktop/files/pen drivers$ dotnet --info
A fatal error occurred. The folder [/usr/share/dotnet/host/fxr] does not exist

this is weird, right?

jamesbt365, 2023-07-30:
You have incorrectly installed dotnet, see the link I sent you above and follow one of the solutions.

xendyex, 2023-07-30:
> You have incorrectly installed dotnet, see the link I sent you above and follow one of the solutions.

Ok.

xendyex, 2023-07-30:
> You have incorrectly installed dotnet, see the link I sent you above and follow one of the solutions.

[image]
Same thing... I'm starting to think something is wrong with my laptop.

xendyex, 2023-07-30:
> You have incorrectly installed dotnet, see the link I sent you above and follow one of the solutions.

*very very very very crazy happy noises*
$ dotnet --info

Host:
  Version:      7.0.9
  Architecture: x64
  Commit:       8e9a17b221

.NET SDKs installed:
  No SDKs were found.

.NET runtimes installed:
  Microsoft.AspNetCore.App 6.0.20 [/usr/lib/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.AspNetCore.App 7.0.9 [/usr/lib/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 6.0.20 [/usr/lib/dotnet/shared/Microsoft.NETCore.App]
  Microsoft.NETCore.App 7.0.9 [/usr/lib/dotnet/shared/Microsoft.NETCore.App]

Other architectures found:
  None

Environment variables:
  Not set

global.json file:
  Not found

Learn more:
  https://aka.ms/dotnet/info

Download .NET:
  https://aka.ms/dotnet/download

it works now :D

closed 2023-07-30 by jamesbt365

xendyex, 2023-07-30:
> You have incorrectly installed dotnet, see the link I sent you above and follow one of the solutions.

Thanks sooooo much!
[image]
Have a great day.