OpenTabletDriver/OpenTabletDriver #2859: Error when launching for the first time "Lost connection to daemon, exiting..."

Lomi0s, 2023-07-30:
### Steps to Reproduce

Open the OpenTabletDriver.UX.Wpf executable

Error message "Lost connection to daemon, exiting..." comes up.

[image]

### Expected Behavior

It should open and I should be able to use it.

### Observed Behavior

Error message and it closes immeditely after.

### Tablet Device

Wacom One

### Diagnostic Information

N/A

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-07-30:
1- Is the Daemon in the same folder as the UX?
2- Could you screenshot the output from the Daemon when you open it?

gonX, 2023-11-19:
Closing issue as stale. Feel free to let us know if you're still having the issue on the latest version of the driver, which is currently 0.6.3.0.

closed 2023-11-19 by gonX