OpenTabletDriver/OpenTabletDriver #2860: Huion H430P not detected

Semizzz-710, 2023-07-31:
### Steps to Reproduce

OS: Windows 11.
When I download the software and open it nothing happens, it says no tablets were detected, so I used the WinUSBCleanup but it said that there were no Gaomon/Huion Drivers, and now i dont know what to do.

### Expected Behavior

It should've detected the tablet after installing but it doesnt for some reason.

### Observed Behavior

Tablet is not detected inside the software.

### Tablet Device

Huion H430P

### Diagnostic Information

[Diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12216527/Diagnostic.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Semizzz-710, 2023-08-01:
When i press dump all nothing happens

Semizzz-710, 2023-08-01:
[Diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12225219/Diag.txt)

jamesbt365, 2023-08-02:
I do **not** need a diagnostic, I need a string dump, the instructions to do that are above, you will need to uninstall the huion driver first.

Semizzz-710, 2023-08-02:
i did a string dump, i just named it diag

jamesbt365, 2023-08-02:
No. Thats not a string dump, that is a diagnostic.

jamesbt365, 2023-08-13:
Going to assume its fixed by #2869.

closed 2023-08-23 by jamesbt365