OpenTabletDriver/OpenTabletDriver #2863: opentabletdriver.service does not exist on tumbleweed

Pizza989, 2023-07-31:
### Steps to Reproduce

1. Install v0.6.2.0 on openSUSE Tumbleweed `sudo zypper in OpenTabletDriver.rpm`
I did not complete this step as of the official install instructions mentioned in the acknoledgements section, but after the description provided in the release. I also manually added the repository for the dotnet runtime following microsofts instructions.
3.  Enable the service `sudo systemctl enable opentabletdriver.service`

### Expected Behavior

Should enable the service. Output would probably be something like created symlink from "some location" to "another location".

### Observed Behavior

Outputs following error message `Failed to enable unit: Unit file opentabletdriver.service does not exist.`

### Tablet Device

Wacom CTL-672

### Diagnostic Information

"N/A"

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-07-31:
Its a user service, not a system one.

Pizza989, 2023-07-31:
how do you enable a user service

Pizza989, 2023-07-31:
nvm got it thx

closed 2023-07-31 by Pizza989