OpenTabletDriver/OpenTabletDriver #2865: Support Request: Unable to detect Gaomon 1060 Pro (256c:0064)

awerty-noob, 2023-08-04:
### What do you need help with?

OS: Archlinux
Tablet: Gaomon 1060 Pro (256c:0064)
No tablets are detected.
I'm sure the tablet is good because I tested it with `digimend-kernel-driver` with 256c:0064 index patched, which runs smoothly except the button on the tablet. So I uninstalled the digimend and installed opentabletdriver from AUR, but "No tablets are detected".

### What have you tried?

Blacklisted hid_uclogic
udev-rules: Installed directly from repo's ./generate-rules.sh and `sudo mv ./bin/99-opentabletdriver.rules /etc/udev/rules.d/99-opentabletdriver.rules` and rebooted.
Configurations:  `cp -r ./OpenTabletDriver.Configurations/Configurations ~/.config/OpenTabletDriver/`

### Diagnostic Information

[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12256503/diagnostic.txt)

And the dump file from device string reader
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12256504/dump.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-08-23:
1 - Please undo the moving on configurations, master configurations don't work properly on 0.6.x.
2 - Try moving [this](https://cdn.discordapp.com/attachments/628651971267788800/1144011915853582528/1060_Pro.json) file into ~/.config/OpenTabletDriver/Configurations and replugging your tablet

If it works, which it should, go into `tablet > tablet debugger`, check the max x and y coordinates and tell me them here. Please also test the max pressure and if the pen buttons go to true when you press them as well as the aux buttons when you press them on the tablet.

awerty-noob, 2023-08-25:
It works well! I think the problem solved.
Max x and y : `<50800,31750>`, and the aux buttons work well.

jamesbt365, 2023-08-26:
> Except the 2 buttons on the pen I can't find the settings of them.
Because I linked you a configuration for 0.6.x, not master, if you want a configuration for master, check the pull request #2899.

awerty-noob, 2023-08-27:
Ok I switch to the master branch, everything works fine!
Closing the issue.

PR #2899 (2023-11-21): Add new string to Gaomon 1060 Pro
supposedly fixes #2865, waiting for the user to confirm at this time. Opening as draft until we receive this.

Adding support with the string `GM001_T216_220814` for tablet `Index 2: Gaomon Tablet_1060Pro`
closed 2023-08-27 by awerty-noob