OpenTabletDriver/OpenTabletDriver #2871: Intuos 3 PTZ-630 bug in Photoshop

shynati, 2023-08-15:
### Steps to Reproduce

While using Adobe Photoshop 2023 or beta I encountered a problem with my Wacom Pen Intuos 3 PTZ-630.

Any additional functions which use pressed mouse and function key, for example Brush resize (Control + Option + drag left/right) or Alt+drag option to copy selected layer, do not work after installation of the OpenTabletDriver.

Original Wacom driver didn't have that issue, but the placement of the mouse did not correspond to the tablet size. It worked as a regular mound, move and drag only when I turned on Adobe Photoshop. it seemed like a glitch.

### Expected Behavior

Use Brush and press Control + Option while dragging, the Brush does not resize, as expected, instead functions as a brush (drawing).

Hold Alt + grab and drag layer, only grabs the layer, copy does not function, only dragging happends.

### Observed Behavior

Additional functions are not working when Alt, Ctrl or other pressed in Photoshop.

### Tablet Device

Wacom Intuos 3 PTZ-630

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.2.0",
  "Build Date": "06/25/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "13.4.0",
    "VersionString": "Unix 13.4.0"
  },
  "Environment Variables": {
    "USER": "shy",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": "/var/folders/t0/3f487g713d73dyhwbnsgwqn40000gn/T/"
  },
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io/AppleT602xIO/apcie@80000000/AppleT6020PCIe/pci-bridge3@3/IOPP/pcie-xhci@0/AppleEmbeddedUSBXHCIASMedia3142@08000000/AppleUSB20XHCIPort@08300000/USB2 Hub@08300000/AppleUSB20InternalHub@08300000/AppleUSB20HubPort@08330000/USB2.0 Hub@08330000/AppleUSB20Hub@08330000/AppleUSB20HubPort@08333000/Keyboard Hub@08333000/AppleUSB20KeyboardHub@08333000/AppleUSB20HubPort@08333200/Apple Keyboard@08333200/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice",
      "Manufacturer": "Apple, Inc",
      "ProductName": "Apple Keyboard",
      "SerialNumber": "",
      "FriendlyName": "Apple Keyboard",
      "VendorID": 1452,
      "ProductID": 545,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "IOService:/IOResources/IOBluetoothHCIController/IOBluetoothDevice/IOBluetoothL2CAPChannel/AppleHSBluetoothDevice",
      "Manufacturer": "Apple",
      "ProductName": "Magic Mouse",
      "SerialNumber": "48-4b-aa-ed-f5-9b",
      "FriendlyName": "Magic Mouse",
      "VendorID": 76,
      "ProductID": 617,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 65,
      "CanOpen": true
    },
    {
      "DevicePath": "IOService:/AppleARMPE/arm-io/AppleT602xIO/apciec0@30000000/AppleT6000PCIeC/pcic0-bridge@0/IOPP/pci-bridge@0/IOPP/pci-bridge@2/IOPP/pci8086,15f0@0/AppleUSBXHCITR@04000000/AppleUSB20XHCIPort@04100000/USB2 Hub@04100000/AppleUSB20Hub@04100000/AppleUSB20HubPort@04140000/Studio Display@04140000/HID Relay@a/AppleUserUSBHostHIDDevice",
      "Manufacturer": "Apple Inc.",
      "ProductName": "Studio Display",
[... 2048 lines cut ...]
    {
      "Time": "2023-08-15T15:25:01.896506+02:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-08-15T15:25:01.896511+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jeffreylm, 2024-02-25:
**Modifiers not working (ALT, Shift and CMD) mac VENTURA 13.6.4**

I am having the exact same issue with my Wacom PTZ-630. Is it possible to design the driver to not need to have to setup a binding? Here is my issue described clearly. And for clarity. I just installed this driver for the first time about an hour ago, so I don't know if it is normal that you have to setup aux to be able to use alt, shift and command keys.

I was expecting the normal way that you simply click and hold the tip pressed and it works as normal.

Here is my issue:

When I assing letters to the aux binding, they work as expected. But when I assign the Alt or LeftAlt, nothing happens. It does not offer the click + Alt option.

I have the Mouse Button Binding (Button:Left) set for tip, and have the Aux binding 1 set to Alt.
When click and drag, you expect it to initiate a copy or duplicate in all apps (OS: VENTURA 13.6.4)

Wacom PTZ-630

sortedcord, 2024-10-02:
Hi, I'm not sure if this is related to the same underlying problem or not, but I havea Gaomon S620 and apparently when I set its buttons to use modifier keys, that also doesn't work. I have set the pen button to use "LeftApplication+LeftShift+P" but on pressing the button, only "P" is inserted and the modifiers are ignored regardless of what application it is. However, if I use the default Gaomon driver and its app then it seems to be working.

OS: MacOS Monterey (12.7.6) (x86_64)
OpenTabletDriver Version: 0.6.4.0

closed 2024-11-25 by InfinityGhost