OpenTabletDriver/OpenTabletDriver #2895: Discussion: Disable Artist Mode smoothing with libinput quirks file

gonX, 2023-08-22:
### Description

By default, libinput smooths all tablet input, even if it is smoothed by OpenTabletDriver already.

I propose disabling the smoothing by adding the following quirks to begin with:

```
[OpenTabletDriver Virtual Tablets]
MatchName=OpenTabletDriver*
AttrTabletSmoothing=0
```

The file path should be `/usr/share/libinput/30-vendor-opentabletdriver.quirks`

---

**Arguments for disabling libinput smoothing by default:**
- Raw(er) input
- Artist mode now handles like absolute mode (as absolute mode is detected as an absolute mouse rather as an tablet device)
- More users are likely interested keeping it disabled, rather than having it enabled.

**Valid arguments against disabling the smoothing provided by libinput:**

- Does not match upstream behavior with other tablets (ie. tablet will feel different from stock driver)
- Can make further configuration (e.g. OpenTabletDriver plugins) mandatory for chattery or otherwise inaccurate tablets, adding to an already somewhat complicated install process

---

Sadly exposing this as an easy user-controlled setting is ambitious as it seems we can only do this in packaging, requiring the user to have root permissions to set local quirk overrides, or root permissions to repackage with their preferred settings..

Down the line I would like to see something like https://github.com/OpenTabletDriver/OpenTabletDriver/pull/1983 being implemented as well, so this would just be a stopgap solution until then (though, the user would still need root). With that in place, users would be able to generate their own quirks file that could re-enable the smoothing if they so desire.

(loosely copied from OpenTabletDriver/OpenTabletDriver.Packaging#22)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

gonX, 2023-10-23:
If there is no opposition I plan on making a pull request that implements this quirks file soon

closed 2025-01-03 by gonX

jtojnar, 2025-06-09:
Upstream fix was proposed in https://gitlab.freedesktop.org/libinput/libinput/-/merge_requests/1219.

PR #3566 (2024-12-14): [0.6.x] Disable libinput smoothing for OpenTabletDriver virtual tablet
As discussed in issue #2895

Fixes #2895 for 0.6.x

Note that packaging maintainers will likely need to update packaging scripts accordingly, unless they use the generic source folder or binary tarball to base their install on.

Check:

- [x] Generic Tarball
- [x] AUR (needs to be reworked to just have OTD "install" instead of copying files manually)
- [x] Debian
- [x] Fedora
- others?
PR #3567 (2025-01-03): Disable libinput smoothing for OpenTabletDriver virtual tablet
Forward port of #3566

Fixes #2895

Linux packagers: notice that this requires you to add a file