OpenTabletDriver/OpenTabletDriver #2900: "Shift" key not working when using Adobe Illustrator or Affinity Designer2

pockygalore, 2023-08-24:
### Steps to Reproduce

When using either Adobe Illustrator and Affinity Designer2, pressing "shift" key when using my wacom to resize an image or a shape will lock its aspect ratio. Meaning the shape or object will maintain its size ratio and will not be distorted no matter how much you stretch it. 

With OpenTabletDriver and my wacom intuos 480 plugged in, the both "shift" keys on my board are disabled in both programs. But holding "Shift" + letter while typing will produce a capital letter in both programs.  

### Expected Behavior

Hold "shift" key when stretching an image, object or shape should force the item to maintain it's size ratio

### Observed Behavior

"shift" + resizing doesn't work.

### Tablet Device

Wacom CTL 480

### Diagnostic Information

NA

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

pockygalore, 2023-08-24:
Forgot to add that when the wacom is unplugged, the "shift" key works fine when resizing.

X9VoiD, 2023-08-24:
Upload diagnostics. Open OTD -> Help -> Export Diagnostics, save the file, then upload it here as text or zip.

pockygalore, 2023-08-24:
[image]
[image]

Please advise, I can't find a Help section. Neither can I find a Diagnostics section.
At best, I can find Console > Info and Console > Debug.  

[image]
[image]

jamesbt365, 2023-08-24:
At the very top of your screen when you are selecting OTD there will be a section called help.

pockygalore, 2023-08-24:
[illustrator and designer2.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12427377/illustrator.and.designer2.json.zip)

Thanks! On further usage with both Illustrator and Designer2, it seems that in any command that requires "shift" + wacom , the "shift" key is disabled. Another example would be drawing a straight line. But using my mousepad with shift works fine. 

I forgot to add that i'm using a Mac with os Monterey 12.6.8

gonX, 2023-10-16:
Have you tried seeing if #2894 fixes it?

gonX, 2026-10-02:
I'm going to assume this is fixed by #2894 (#3544) - v0.6.5 onwards.

closed 2026-10-02 by gonX