OpenTabletDriver/OpenTabletDriver #2911: Support Request: Cursor no longer moving when using artist mode linux

nicolas3121, 2023-09-02:
### What do you need help with?

everything seems to work in absolute mode but once i switch to artist mode the cursor no longer moves and buttons no longer seem to be detected.
Everything does look fine in the tablet debugger. 
The pen tip setting has been cleared and buttons mapped to artist mode ones
No errors are shown in the console.

### What have you tried?

- reloaded udev rules and restarted computer.
- checked settings and looked for errors in console

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.2.0",
  "Build Date": "9/2/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.4.12.1",
    "VersionString": "Unix 6.4.12.1"
  },
  "Environment Variables": {
    "USER": "nicolas",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/home/nicolas",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/bin:/home/nicolas/.dotnet/tools:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:03:00.4/usb3/3-2/3-2:1.0/0003:056A:0376.0001/hidraw/hidraw0",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos BT S",
      "SerialNumber": "1AH00S2028490",
      "FriendlyName": "Intuos BT S",
      "VendorID": 1386,
      "ProductID": 886,
      "InputReportLength": 192,
      "OutputReportLength": 0,
      "FeatureReportLength": 1281,
      "CanOpen": true
    }
  ],
  "Console Log": [
    {
      "Time": "2023-09-02T19:46:24.5702129+02:00",
      "Group": "Plugin",
      "Message": "Loading plugin 'Windows Ink'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-09-02T19:46:24.5703072+02:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-02T19:46:24.5703338+02:00",
      "Group": "Detect",
      "Message": "Searching for tablet '10moon 1060N'",
      "StackTrace": null,
[... 1915 lines cut ...]
    {
      "Time": "2023-09-02T19:46:24.5757379+02:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-09-02T19:46:24.5757597+02:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

X9VoiD, 2023-09-02:
Sorry but this isn't something that can be fixed within OTD itself. OTD re-exposes the data as a "virtual tablet" that acts like a "real tablet" to Linux. If your DE/WM can't handle a tablet input then there's nothing we can do about that.

closed 2023-09-02 by X9VoiD

nicolas3121, 2025-12-07:
I was trying out the wayland version of qtile back then which sadly still hasn't implemented the relevant protocol i think