OpenTabletDriver/OpenTabletDriver #2912: Won't detect tablet (Deco LW)

BurritoKitten, 2023-09-03:
### What do you need help with?

I can't get OpenTabletDriver to detect my Deco LW. 

### What have you tried?

I've tried uninstalling/reinstalling the given driver for the device, and that did not fix the problem.
Whenever I try "Detect tablet", it says "No tablets were detected"

### Diagnostic Information

[Help.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12504940/Help.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Beryesa, 2023-10-01:
@BurritoKitten Can I ask does it work at all with the default wacom driver, or are you stuck with the proprietary one?

jamesbt365, 2023-10-01:
Given as this is not a Wacom tablet, no it will not work with the Wacom driver.

Beryesa, 2023-10-07:
> Given as this is not a Wacom tablet, no it will not work with the Wacom driver.

Sorry I thought they were using linux before asking so.

gonX, 2023-10-08:
The Deco LW is not currently on our supported list. This needs a new configuration.

maokwen, 2024-08-26:
It seems that Deco LW has the same `ProductID` as Deco L, but `InputReportLength` changed from `10` to `14`.

Here is my configuration file (working for wired):
```json
{
  "Name": "XP-Pen Deco LW",
  "Specifications": {
    "Digitizer": {
      "Width": 254,
      "Height": 152.4,
      "MaxX": 50800,
      "MaxY": 30480
    },
    "Pen": {
      "MaxPressure": 8191,
      "ButtonCount": 2
    },
    "AuxiliaryButtons": {
      "ButtonCount": 8
    },
    "MouseButtons": null,
    "Touch": null
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 10429,
      "ProductID": 2357,
      "InputReportLength": 14,
      "OutputReportLength": 33,
      "ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenOffsetPressureReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": [
        "ArAE"
      ],
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxiliaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

But I can't get wireless configuration work.

gonX, 2025-10-06:
@maokwen Feel free to pull request this configuration to our `0.6.x` branch. You may need to clean up the configuration a bit to pass our tests, but the failing test output should help you clean it up.