OpenTabletDriver/OpenTabletDriver #2913: Support Request:

intelpel, 2023-09-03:
### What do you need help with?

Hello! Opentablet can not detect ParbloN4. 
_All details of issue and scrennshoots you can find at my tred:_
(https://osu.ppy.sh/community/forums/topics/1816691?n=1)
**Thank you!**

### What have you tried?

update opentablet
deleted old drivers
Zadig's WinUSB installed
Download [this](https://cdn.discordapp.com/attachments/628651971267788800/1144013661191553235/A640.json) file, create a folder called Configurations next to the executables of OpenTabletDriver (OpenTabletDriver.Daemon.exe, OpenTabletDriver.UX.Wpf.exe, etc, etc) and place this file into this folder

### Diagnostic Information

[123.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12505963/123.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-03:
wait until the next release of OpenTabletDriver.

Also next time you make an issue, check for other issues in the issues tab, which you make your issue from, and title your issue accordingly instead of blazing past all the boxes that are there to help you. Thank you!

jamesbt365, 2023-09-03:
Oh and, well if you're going to do configurations override don't install winusb for tablets that don't require it. Run TabletDriverCleanup.

closed 2023-09-03 by jamesbt365