OpenTabletDriver/OpenTabletDriver #2914: Add support for Artisul D16

googlyeyesultra, 2023-09-03:
### Manufacturer

Artisul

### Product Name

Artisul D16

### Device Width

344.16mm

### Device Height

193.59mm

### Pressure Levels

8192

### Lines per Inch (LPI)

Not sure

### Additional Information

8 buttons and a scroll wheel on the tablet. 2 buttons on the pen.

Windows PNP drivers work, but do not support the tablet buttons.

Someone wrote support for it for digimend driver on Linux, in case something from there will be helpful. https://github.com/DIGImend/digimend-kernel-drivers/pull/483

Diagnostics and string export are both attached.

[string dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12506700/string.dump.txt)

### Diagnostic Information

[artisuld16_diagnostics.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12506705/artisuld16_diagnostics.json.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-03:
Configurations are far easier to support over discord, if you are in the discord and it has gone stale, bump your issue. Otherwise join our discord [Discord](https://opentabletdriver.net/discord)

googlyeyesultra, 2023-09-26:
Uploading wireshark pcap here just so everything is in one place. OEM driver installed, start capture, mark ignored everything that pops up there, then plugged in tablet. No buttons/pen use, just initialization.

[initialization.pcapng.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12730403/initialization.pcapng.txt)