OpenTabletDriver/OpenTabletDriver #2916: Support Request: Gaomon 1060 Pro cannot connect to OpenTabletDriver

questioninganxiety, 2023-09-04:
### What do you need help with?

So I just got my hands on the Gaomon 1060Pro and I tried to connect it to OpenTabletDriver but it didn't work. I have never used a tablet therefore I didn't download any kinds of drivers except window ones. 

### What have you tried?

I tried fixing it for hours but for some reasons my OpenTabletDriver didn't have a configuration, which was the solution for the most of the problems I managed to search.
[image]

### Diagnostic Information

[diagno.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12513621/diagno.zip)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-04:
fixed in #2899.

closed 2023-09-04 by jamesbt365

questioninganxiety, 2023-09-04:
Can you show me the specific steps? I think I'm too old to catch up with those instructions. Thanks

jamesbt365, 2023-09-04:
Download [this](https://cdn.discordapp.com/attachments/615611007951306863/1147208890933588058/1060_Pro.json), create a folder called `Configurations` next to the executables of OTD (OpenTabletDriver.Daemon.exe & OpenTabletDriver.UX.Wpf.exe) and place that file inside the folder.

questioninganxiety, 2023-09-04:
Oh man it worked! Thanks a lot! Do I need to send proof or something to you?

jamesbt365, 2023-09-04:
Nope.