OpenTabletDriver/OpenTabletDriver #2922: Support Request:

OxugenRed, 2023-09-06:
### What do you need help with?

Parblo ninos n7 no tablet detected 

### What have you tried?

I waited for the new driver version, but my tablet is still not detected

### Diagnostic Information

nor working 

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-06:
Run [TabletDriverCleanup](https://github.com/X9VoiD/TabletDriverCleanup/releases/latest) and actually fill out the templates next time instead of disregarding the information that is meant to help both of us out! Thanks for not following any of the information or instructions, or giving your issue a proper title. 

Next time you open an issue help us help you by actually following the instructions provided.

closed 2023-09-06 by jamesbt365

OxugenRed, 2023-09-06:
> Run [TabletDriverCleanup](https://github.com/X9VoiD/TabletDriverCleanup/releases/latest) and actually fill out the templates next time instead of disregarding the information that is meant to help [...]

it really helped, thanks for the help. Next time I'll do everything according to the instructions 😘💋