OpenTabletDriver/OpenTabletDriver #2925: Not working on Nix os

View-my-Git-Lab-krafi, 2023-09-09:
### Steps to Reproduce

at first time i open that file
`
 sudo nano /etc/nixos/configuration.nix
`
then i simply added this line on environment.systemPackages = with pkgs;
`
  # List packages installed in system profile. To search, run:
  environment.systemPackages = with pkgs; [
   # opentabletdriver
   # veikk-linux-driver-gui
  ];

`
sudo nixos-rebuild switch

now I can see the opentabletdriver icon but if i open after about 10 sec it shows an error  like "  **make sure the deamon is running**  " 
then i read the official wiki is says just add  those lines 
`
# Enable OpenTabletDriver
   hardware.opentabletdriver.enable = true;
`
still same issue , what i tried is at my own 
`systemctl --user start opentabletdriver
`
now the setup wizard run successfully but the issue is it coudnt find my tablet vk640

here is the  console

`[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw4.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw2.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw4.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw2.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
[... 1813 lines cut ...]
    {
      "Time": "2023-09-10T00:14:00.3834523+06:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-09-10T00:14:00.383457+06:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-09:
Reboot, udev doesnt apply instantly. Not really a bug of any sort. Though I'm assuming packaging changes can be done to mitigate the needs to reboot on NixOS.

If for whatever reason udev still doesnt apply you may need to run the generate rules script in our repo and look at the nixos documentation in to see how to do that.

I do have an example on my NixOS config on my github ~~though its messy and needs work~~ if youre still confused.

closed 2023-09-09 by jamesbt365

View-my-Git-Lab-krafi, 2023-09-10:
```
   hardware.opentabletdriver.enable = true;
   services.udev.extraRules = "";
```
adding this 2 line on  /etc/nixos/configuration.nix  

 then switch to the unstable branch I did the following:
```

sudo nix-channel --add https://channels.nixos.org/nixos-unstable nixos

sudo nix-channel --update

sudo nixos-rebuild switch 
```

it solved the issue on nixos please add this on the documentation .

jamesbt365, 2023-09-10:
Nothing to add to our documentation. You didnt add any extra rules, you didn't need to add any extra rules. You would only need to do that with unsupported tablets.

Its common in NixOS to switch channel for specific packages or systemwide.

The only "documentation" needed is for NixOS, which has documentated channel switching already.

View-my-Git-Lab-krafi, 2023-09-12:
hay is it possible to use nixos wayland hyprland with opentabletdriver ?

jamesbt365, 2023-09-12:
Hyprland doesnt have any support for tablets, so support here is flaky. Your mileage may vary.