OpenTabletDriver/OpenTabletDriver #2926: Xp-Pen Tablet Deco Mw Support Request

snow-nene, 2023-09-11:
### Description

I want to use this driver for Deco Mw. But it doesn't support it yet.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-11:
Either reopen your issue under the configuration request template or fill out the information on that template here.

The information you have provided is not enough to create a configuration.

closed 2023-09-11 by jamesbt365

Beryesa, 2023-10-01:
@snow-nene Can I ask does it work at all with the default wacom driver? Or are you stuck with the proprietary one?

jamesbt365, 2023-10-01:
It is not a Wacom tablet, so it will not work with the Wacom driver.

Beryesa, 2023-10-07:
> It is not a Wacom tablet, so it will not work with the Wacom driver.

Sorry I thought they were on Linux for some reason, forgot otd is cross-platform lol
That statement isn't correct over here