OpenTabletDriver/OpenTabletDriver #2930: issues with X and Y scaling with 90degree rotation of tablet

m10d, 2023-09-17:
### Steps to Reproduce

connect a graphics tablet to your PC.  Follow steps 1-6 (see detailed annotated graphics below for what I see).  For all we're connecting the tablet as a 2nd monitor, "extending" the desktop (vs mirroring, or overlaying/overlapping the tablet on the existing screen(s)

<saying comment too long, see in the pics>

### Expected Behavior

I should be able to easily set my tablet up in portrait, just as it works in landscape in bullets 1-3

### Observed Behavior

for completeness, these are as expected:
[image]

but then it all falls apart in portrait:
[image]

### Tablet Device

huion kamvas 12 over usb-c alt mode

### Diagnostic Information

says too long :(

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

m10d, 2023-09-17:
diagnostic info:
{
  "App Version": "OpenTabletDriver v0.6.3.0",
  "Build Date": "09/06/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.2.0.32",
    "VersionString": "Unix 6.2.0.32"
  },
  "Environment Variables": {
    "USER": "mike",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/home/mike",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/mike/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:32AC:0002.0003/hidraw/hidraw2",
      "Manufacturer": "Framework",
      "ProductName": "HDMI Expansion Card",
      "SerialNumber": "11AD1D0005D63E0122200B00",
      "FriendlyName": "HDMI Expansion Card",
      "VendorID": 12972,
      "ProductID": 2,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:256C:006D.0008/hidraw/hidraw4",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 12,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:256C:006D.0009/hidraw/hidraw5",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 28,
      "OutputReportLength": 0,
      "FeatureReportLength": 257,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.2/0003:256C:006D.000A/hidraw/hidraw7",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 8,
      "OutputReportLength": 0,
[... 2430 lines cut ...]
      "Time": "2023-09-16T21:44:21.7723542-06:00",
      "Group": "Huion Kamvas 12",
      "Message": "Tip Binding: [Mouse Button Binding: Left]@0%",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-16T21:44:21.7723572-06:00",
      "Group": "Huion Kamvas 12",
      "Message": "Filters: OpenTabletDriver.Desktop.Binding.BindingHandler",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-16T21:44:21.7723602-06:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

X9VoiD, 2023-09-17:
Do not use your DE's settings for tablets, those interfere with OTD's mapping as you have pointed out and there's no way for OTD to correct for such changes. Do everything inside OTD.

closed 2023-09-17 by X9VoiD

m10d, 2023-09-17:
respectfully @X9VoiD this is not true - OTD accepts the monitor layout as provided (and configured by) the desktop environment.  The DE does this, in **all** cases, prior to OTD being involved in anyway.  This is why OTD displays the (DE configured) monitor layout in the upper pane.

In my case, the only config I have done is setting how my monitors are physically laid out.  Of course this step is unavoidable.  And the only reason I include anything about DE configuration is to demonstrate what *is* working.

If you'd like, just pretend I opened the bug at step6 diretly: here is my screen, there's a bug, or there is something I am doing wrong in OTD.  Also, per directions/FAQ, all the data in this issue takes place after following the FAQ, including blacklisting `wacom`.  I've gone ahead and reproduced this without any of the DE interaction that seems to be at issue:

1. my pc is set up
2. reboot PC
3. open and attempt to configure OTD.  Scaling bug is observed:

[image]

m10d, 2023-09-17:
i think i have narrowed down the bug by experimentation.  

`lock to usable area` and `width`/`height` are not used by the UI/OTD-config-app in the same coordinate system.  

Lock to usable area incorrectly truncates the width/height's entered.  The height field displays the intended value, but the visual (and apparently the model that's used internally) has truncated it.

Shall I submit a new bug that zeros in on this, or can this bug work for the team?

m10d, 2023-09-17:
here's how I found/worked-around this bug:

1) noticed the inconsistency in height
[image]

2) specifically, in order to get the `height` I have set (267.8mm) applied, even after run-checking `lock to usable area`, I have to edit the field (delete any char, re-add same char)
[image]

3) finally, after doing this, this config works (cursor follows pen position precisely across the actual tablet area)
[image]

m10d, 2023-09-17:
`artist mode` was brought up several times - I did not experience any difference in behavior with or without `artist` vs `absolute` in this issue.  The one change I did notice is that I couldn't really use the pen as a pen without using `artist mode` (could not draw continuous lines with pen in a drawing app, in any tablet rotation, until artist-mode was set

m10d, 2023-09-17:
finally, to summarize what a fix would accomplish for this:

1) `lock to usable area` in the `Tablet` section of the OTD config should respect rotation.  Today it does not - but the user is still required to set width/height  _post_ rotation (eg set H>W if in portrait mode), ie these two are inconsistent.  TBC, 
    - ideally, the Tablet visualization could also adjust for rotation, it's very confusing as-is.  But this is less important than the actual settings being consistent
2) enforce all limits on the text fields `width`|`height` etc.  Today, in the circumstances I described above, I can set them but they are incorrect.  this is also related to lock-to-available.  That check box must do exactly ONE of these, not a mix:
     a) ignore w/h if it's outside of the bounds --> but then accept them when it's unchecked (today when unchecked it doesn't do anything, as I saw when I had to edit/re-add the same number for height that was already being displayed)
    b) or it could just truncate/update the number in (my case `height` field) to what it is actually using.  Then the _user_ has to update the field to the larger number needed _after_ un-checking lock-to-usable.

X9VoiD, 2023-09-17:
Duplicate of #1143

m10d, 2023-09-18:
> Duplicate of #1143

Folks - no, it's not a duplicate of #1143 which deals with changes in monitor layout (eg DE, above/outside OTD)

**My issue is 100% within OTD**, and it requires ZERO interactions with DE-level settings, or monitor changes.

Obviously my OP has some DE monitor config - because _at that time I obviously had no idea where the bug was, and I was trying to show what worked, what seemed wrong ..._  (and all quite painstakingly broken down / summized / screenshoted to make it digestible by the team)

Again - this bug has nothing to do with DE changes. To quote https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2930#issuecomment-1722553019:

>  1) my pc is set up
>  2) reboot PC    _<-----  it's a clean boot.  **NO** DE interaction!!_
>  3) open and attempt to configure OTD. Scaling bug is observed:

Shall I just open a new bug starting at https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2930#issuecomment-1722558111?

m10d, 2023-09-18:
note to self: https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h  keycodes (not related to this issue)

gonX, 2023-10-01:
Sorry, the issue is really really long and difficult to keep track of what the issue actually is. I tried setting up a rotated monitor and saving settings for that, and it seems to perfectly center when starting OTD.

Here is what OTD's settings looks like for me:
[image]

This smells more like a combination of #2573 #2594 than anything else, as well as using your DE's settings. If you can reproduce this in a default WM it is more likely your bug is able to be reproduced.