OpenTabletDriver/OpenTabletDriver #2931: null ref exception when restarting daemon

m10d, 2023-09-17:
### Steps to Reproduce

1. reboot PC (just to be clear, no DE changes were involved in between the OTD steps), log in.  Observe my tablet (as a monitor) is set up as I expect it.
2. open OTD app, change some settings while triaging https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2930
3. close OTD app, then `systemctl --user restart opentabletdriver.service`
4. open OTD app again

### Expected Behavior

no error

### Observed Behavior

[image]

persists (I cannot open OTD app) until I reboot (did not try ending session, just rebooted)

### Tablet Device

probably N/A but huion12

### Diagnostic Information

```
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

m10d, 2023-09-17:
here's my diagnostic info (github said it was too long to include in the issue).  But note this was gathered after I rebooted / didn't see error any longer, and it may include a bunch of stuff from triaging the earlier issue. 

Just hoping by submitting this issue this null-ref can be found, I didn't do anything besides update settings in in OTD, so hopefully it's not some esoteric code path.

```
{
  "App Version": "OpenTabletDriver v0.6.3.0",
  "Build Date": "09/06/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.2.0.32",
    "VersionString": "Unix 6.2.0.32"
  },
  "Environment Variables": {
    "USER": "mike",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/home/mike",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/mike/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:256C:006D.0004/hidraw/hidraw3",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 28,
      "OutputReportLength": 0,
      "FeatureReportLength": 257,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.1/0003:32AC:0002.0006/hidraw/hidraw5",
      "Manufacturer": "Framework",
      "ProductName": "HDMI Expansion Card",
      "SerialNumber": "11AD1D0005D63E0122200B00",
      "FriendlyName": "HDMI Expansion Card",
      "VendorID": 12972,
      "ProductID": 2,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006D.0005/hidraw/hidraw4",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
      "VendorID": 9580,
      "ProductID": 109,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:256C:006D.0003/hidraw/hidraw2",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_GS1161",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_GS1161",
[... 2738 lines cut ...]
      "Time": "2023-09-17T14:29:56.5118685-06:00",
      "Group": "Huion Kamvas 12",
      "Message": "Tip Binding: [Mouse Button Binding: Left]@0%",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-17T14:29:56.5118716-06:00",
      "Group": "Huion Kamvas 12",
      "Message": "Filters: OpenTabletDriver.Desktop.Binding.BindingHandler",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-17T14:29:56.5118749-06:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}```

m10d, 2023-09-18:
reproduced this a different way.  This path is obviously less valuable than the OP, but I figured I should `strace` it anyway.  

I don't believe there'd be is any sensitive in here, especially this early in the process, but if there might be please let me know, I have not attempted to decode any strings.

```
execve("/usr/bin/otd-gui", ["otd-gui"], 0x7ffceba70ae0 /* 46 vars */) = 0
brk(NULL)                               = 0x555adc8bb000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffcb9f043e0) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8ddf53a000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=81179, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 81179, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8ddf526000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\237\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\"\233}\305\t\5?\344\337^)\350b\231\21\360"..., 68, 896) = 68
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=2216304, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2260560, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8ddf200000
mmap(0x7f8ddf228000, 1658880, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7f8ddf228000
mmap(0x7f8ddf3bd000, 360448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1bd000) = 0x7f8ddf3bd000
mmap(0x7f8ddf415000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x214000) = 0x7f8ddf415000
mmap(0x7f8ddf41b000, 52816, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f8ddf41b000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8ddf523000
arch_prctl(ARCH_SET_FS, 0x7f8ddf523740) = 0
set_tid_address(0x7f8ddf523a10)         = 102003
set_robust_list(0x7f8ddf523a20, 24)     = 0
rseq(0x7f8ddf5240e0, 0x20, 0, 0x53053053) = 0
mprotect(0x7f8ddf415000, 16384, PROT_READ) = 0
mprotect(0x555adb4b9000, 4096, PROT_READ) = 0
mprotect(0x7f8ddf574000, 8192, PROT_READ) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
munmap(0x7f8ddf526000, 81179)           = 0
getrandom("\xf5\xd2\x0f\x31\xa9\x21\xd1\xff", 8, GRND_NONBLOCK) = 8
brk(NULL)                               = 0x555adc8bb000
brk(0x555adc8dc000)                     = 0x555adc8dc000
openat(AT_FDCWD, "/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=5712208, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 5712208, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8ddec00000
close(3)                                = 0
execve("~/scripts/sh", ["sh", "/usr/bin/otd-gui"], 0x7ffcb9f045c8 /* 46 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/local/sbin/sh", ["sh", "/usr/bin/otd-gui"], 0x7ffcb9f045c8 /* 46 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/local/bin/sh", ["sh", "/usr/bin/otd-gui"], 0x7ffcb9f045c8 /* 46 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/sbin/sh", ["sh", "/usr/bin/otd-gui"], 0x7ffcb9f045c8 /* 46 vars */) = -1 ENOENT (No such file or directory)
execve("/usr/bin/sh", ["sh", "/usr/bin/otd-gui"], 0x7ffcb9f045c8 /* 46 vars */) = 0
brk(NULL)                               = 0x55ba5c6e5000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffe23c578f0) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f961f0f0000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=81179, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 81179, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f961f0dc000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\237\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\"\233}\305\t\5?\344\337^)\350b\231\21\360"..., 68, 896) = 68
newfstatat(3, "", {st_mode=S_IFREG|0755, st_size=2216304, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2260560, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f961ee00000
mmap(0x7f961ee28000, 1658880, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7f961ee28000
mmap(0x7f961efbd000, 360448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1bd000) = 0x7f961efbd000
mmap(0x7f961f015000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x214000) = 0x7f961f015000
mmap(0x7f961f01b000, 52816, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f961f01b000
close(3)                                = 0
[... cut ...]
vfork()                                 = 102004
rt_sigprocmask(SIG_SETMASK, [], ~[KILL STOP RTMIN RT_1], 8) = 0
wait4(-1, [{WIFEXITED(s) && WEXITSTATUS(s) == 0}], 0, NULL) = 102004
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=102004, si_uid=1000, si_status=0, si_utime=4, si_stime=1} ---
rt_sigreturn({mask=[]})                 = 102004
wait4(-1, 0x7ffe23c5753c, WNOHANG, NULL) = -1 ECHILD (No child processes)
read(10, "", 8192)                      = 0
exit_group(0)                           = ?
+++ exited with 0 +++

```

gonX, 2023-10-16:
The `strace` doesn't really help (for me, anyway) since its being run on the shell script. Try running `strace` directly on the executable, e.g. `/usr/lib/opentabletdriver/OpenTabletDriver.UX.Gtk`.

With the current reproduction steps provided I cannot reproduce the issue; launch gui, change some things, save, close gui, restart opentabletdriver daemon, start gui - works fine.

The initial stack trace you provided seems to indicate the null ref being within Eto, too, so it would be hard for us to track down and fix the issue.

You can try resetting the driver settings by stopping the opentabletdriver service, deleting your OpenTabletDriver folder (~/.config/OpenTabletDriver` and then starting the service again.

Closing as `wontfix` (for now) since the reproduction steps does not work. You are welcome to reopen the issue if you find other reproduction steps.

closed 2023-10-16 by gonX