OpenTabletDriver/OpenTabletDriver #2934: Support Request: Cursor teleporting on keyboard presses

FlySlime, 2023-09-25:
### What do you need help with?

When using Artist Mode the cursor freaks out and starts teleporting. I notice that it only "teleports" when pressing a key on my keyboard, same issue as mentioned in https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2561#issue-1512958769.

### What have you tried?

I've ensured that the built-in wacom driver is turned off.

``blacklist wacom`` is included in ``/etc/modprobe.d/blacklist``
``lsmod | grep wacom`` returns nothing.

I've also tried resetting to defaults in OTD, as mentioned in https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2234#issuecomment-1120517215.
            
What's going on?

I'm using GNOME 44 Wayland, 6.5.4-zen2-1-zen.

### Diagnostic Information

```
[Wacom CTL-472:Info]	Output mode: Artist Mode
[Wacom CTL-472:Info]	Display area: [3440x1440@<1720, 720>:0°],
[Wacom CTL-472:Info]	Tablet area: [60x25.116278@<76, 47.5>:0°],
[Wacom CTL-472:Info]	Clipping: Enabled
[Wacom CTL-472:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-472:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Wacom CTL-472:Info]	Filters: RadialFollow.RadialFollowSmoothingTabletSpace, OpenTabletDriver.Desktop.Binding.BindingHandler
[Settings:Info]	Driver is enabled.
[Wacom CTL-472:Info]	Output mode: Artist Mode
[Wacom CTL-472:Info]	Display area: [3440x1440@<1720, 720>:0°],
[Wacom CTL-472:Info]	Tablet area: [60x25.116278@<76, 47.5>:0°],
[Wacom CTL-472:Info]	Clipping: Enabled
[Wacom CTL-472:Info]	Ignoring reports outside area: Disabled
[Wacom CTL-472:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Wacom CTL-472:Info]	Filters: RadialFollow.RadialFollowSmoothingTabletSpace, OpenTabletDriver.Desktop.Binding.BindingHandler
[Settings:Info]	Driver is enabled.
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-25:
Could you get a recording? Though if your cursor is teleporting when you're pressing keys on the keyboard this is highly unlikely to be something related to OTD.

FlySlime, 2023-09-25:
Sure, I'll grab a recording of it shortly. If left unattended the cursor stars flickering on its own, from what I've seen this seems to be independent of the key-presses?

FlySlime, 2023-09-25:
Here is a recording showcasing the issue:

https://github.com/OpenTabletDriver/OpenTabletDriver/assets/50276341/90ceee75-528c-4831-bf98-f1b6ab5f8d43

It begins with me showcasing that Absolute Mode works. Afterwards I change to Artist Mode, in which immediatily the cursor starts flickering (on my side), however on the recording the cursor just flat out disappears. When I start writing in the terminal, the cursor appears on the "Save" button, the time when I applied Artist Mode.

FlySlime, 2023-09-26:
Tried it out on KDE Wayland, and the issue persists. The cursor starts flickering when applying Artist Mode. However, in KDE the cursor doesn't teleport when typing anything.

gonX, 2023-09-30:
Does `libinput debug-events` report multiple endpoints (not the same `eventXX`) when you move your pen over the tablet? Normal artist mode behavior mode should look similar to:
```
-event22  TABLET_TOOL_PROXIMITY   +1.783s		13.41*/5.28*	tilt: -3.02*/-3.02*	pressure: 0.00*	pen      (0, id 0) proximity-in 	axes:pt	btn:SS2
 event22  TABLET_TOOL_AXIS        +1.788s		13.38*/5.34*	tilt: -3.02/-2.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.793s		13.34*/5.40*	tilt: -3.02/-2.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.798s		13.30*/5.45*	tilt: -3.02/-1.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.803s		13.26*/5.50*	tilt: -4.02*/-1.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.808s		13.21*/5.56*	tilt: -4.02/0.00*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.813s		13.16*/5.62*	tilt: -3.02*/0.00	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.818s		13.12*/5.68*	tilt: -3.02/0.00	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.823s		13.07*/5.73*	tilt: -3.02/0.00	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.828s		13.02*/5.80*	tilt: -2.01*/1.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.833s		12.97*/5.85*	tilt: -2.01/1.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.838s		12.94*/5.86*	tilt: -1.01*/1.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.843s		12.90*/5.95*	tilt: 0.00*/1.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.848s		12.86*/5.97*	tilt: 0.00/0.00*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.853s		12.82*/6.01*	tilt: 1.01*/0.00	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.858s		12.79*/6.04*	tilt: 1.01/-1.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.863s		12.75*/6.07*	tilt: 1.01/-1.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.869s		12.72*/6.08*	tilt: 1.01/-2.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.873s		12.67*/6.12*	tilt: 1.01/-2.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.878s		12.63*/6.13*	tilt: 1.01/-3.02*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.883s		12.59*/6.15*	tilt: 0.00*/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.888s		12.55*/6.18*	tilt: 0.00/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.893s		12.51*/6.21*	tilt: 0.00/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.898s		12.48*/6.25*	tilt: 1.01*/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.903s		12.44*/6.30*	tilt: 2.01*/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.908s		12.40*/6.35*	tilt: 3.02*/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.913s		12.37*/6.41*	tilt: 5.03*/-3.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.918s		12.33*/6.48*	tilt: 6.03*/-4.02*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.923s		12.30*/6.54*	tilt: 8.04*/-4.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.928s		12.26*/6.61*	tilt: 10.05*/-4.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.934s		12.23*/6.67*	tilt: 12.06*/-4.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.939s		12.19*/6.73*	tilt: 14.07*/-5.03*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.944s		12.15*/6.78*	tilt: 15.08*/-5.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.949s		12.11*/6.82*	tilt: 16.08*/-6.03*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.954s		12.08*/6.86*	tilt: 17.09*/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.959s		12.04*/6.89*	tilt: 17.09/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.964s		12.01*/6.92*	tilt: 17.09/-7.04*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.969s		11.97*/6.94*	tilt: 16.08*/-7.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.974s		11.94*/6.95*	tilt: 16.08/-7.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.979s		11.92*/6.96*	tilt: 15.08*/-8.04*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.984s		11.90*/6.97*	tilt: 15.08/-8.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.989s		11.88*/6.97*	tilt: 14.07*/-8.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.994s		11.87*/6.96*	tilt: 14.07/-8.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +1.999s		11.87*/6.94*	tilt: 13.07*/-7.04*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.004s		11.87*/6.92*	tilt: 12.06*/-7.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.009s		11.88*/6.91*	tilt: 11.06*/-7.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.014s		11.90*/6.89*	tilt: 11.06/-7.04	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.019s		11.92*/6.88*	tilt: 10.05*/-6.03*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.024s		11.94*/6.86*	tilt: 10.05/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.029s		11.96*/6.84*	tilt: 9.05*/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.034s		11.99*/6.84*	tilt: 9.05/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.039s		12.01*/6.82*	tilt: 9.05/-6.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.044s		12.03*/6.82*	tilt: 8.04*/-5.03*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.049s		12.05*/6.81*	tilt: 8.04/-5.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.054s		12.08*/6.82*	tilt: 7.04*/-5.03	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.059s		12.11*/6.83*	tilt: 6.03*/-4.02*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.064s		12.13*/6.86*	tilt: 6.03/-4.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.069s		12.17*/6.89*	tilt: 5.03*/-4.02	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.074s		12.21*/6.94*	tilt: 4.02*/-4.02	pressure: 0.00
[... cut ...]
 event22  TABLET_TOOL_AXIS        +2.099s		12.53*/7.45*	tilt: -1.01*/-2.01	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.105s		12.61*/7.59*	tilt: -1.01/-1.01*	pressure: 0.00
 event22  TABLET_TOOL_AXIS        +2.109s		12.69*/7.75*	tilt: -2.01*/-1.01	pressure: 0.00
 event22  TABLET_TOOL_PROXIMITY   +2.114s		12.69/7.75	tilt: -2.01/-1.01	pressure: 0.00	pen      (0, id 0) proximity-out 
```

Similarly, you should only be seeing 1 tablet device with `libinput list-devices` (can use `libinput list-devices | grep -E 'Cap.*tablet'` for quick look).

FlySlime, 2023-10-02:
`libinput debug-events` returns:

```
event5  - Wacom Co.,Ltd. CTL-472 Mouse: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
-event1   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event0   DEVICE_ADDED            Power Button                      seat0 default group2  cap:k
-event6   DEVICE_ADDED            Logitech USB Receiver             seat0 default group3  cap:p left scroll-nat scroll-button
-event7   DEVICE_ADDED            Logitech USB Receiver Keyboard    seat0 default group3  cap:kp scroll-nat
-event8   DEVICE_ADDED            Kingston HyperX Cloud II Wireless Consumer Control seat0 default group4  cap:k
-event14  DEVICE_ADDED            Sony Interactive Entertainment Wireless Controller Touchpad seat0 default group5  cap:pg  size 71x51mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on dwtp-on
-event2   DEVICE_ADDED            SayoDevice SayoDevice M3K RGB     seat0 default group6  cap:k
-event3   DEVICE_ADDED            SayoDevice SayoDevice M3K RGB Mouse seat0 default group6  cap:p left scroll-nat scroll-button
-event4   DEVICE_ADDED            SayoDevice SayoDevice M3K RGB Consumer Control seat0 default group6  cap:kp scroll-nat
-event10  DEVICE_ADDED            Logitech Wireless Mouse PID:407f Keyboard seat0 default group7  cap:k
-event11  DEVICE_ADDED            Logitech Wireless Mouse PID:407f Mouse seat0 default group7  cap:p left scroll-nat scroll-button
-event16  DEVICE_ADDED            HD 1080P  PC-Camera: HD 1080P     seat0 default group8  cap:k
-event30  DEVICE_ADDED            OpenTabletDriver Virtual Artist Tablet seat0 default group9  cap:T  size 54x14mm
 event30  TABLET_TOOL_PROXIMITY   +0.000s		0.00/0.00	tilt: 0.00/0.00	pressure: 0.00	pen      (0, id 0) proximity-in 	axes:pt	btn:SS2
 event30  TABLET_TOOL_PROXIMITY   +0.000s		0.00/0.00	tilt: 0.00/0.00	pressure: 0.00	pen      (0, id 0) proximity-out 
 event30  TABLET_TOOL_PROXIMITY   +1.146s		22.12*/10.75*	tilt: 0.00*/0.00*	pressure: 0.00*	pen      (0, id 0) proximity-in 	axes:pt	btn:SS2
 event30  TABLET_TOOL_AXIS        +1.152s		22.11*/10.75*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.160s		22.08*/10.76*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.168s		22.01*/10.80*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.176s		21.90*/10.86*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.182s		21.77*/10.95*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.190s		21.63*/11.05*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.198s		21.51*/11.17*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.204s		21.41*/11.30*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.212s		21.34*/11.42*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.220s		21.29*/11.52*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.228s		21.27*/11.60*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.234s		21.28*/11.65*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.242s		21.32*/11.68*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.250s		21.40*/11.66*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.258s		21.55*/11.61*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.266s		21.75*/11.54*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.272s		21.98*/11.46*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.280s		22.24*/11.40*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.288s		22.49*/11.39*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.294s		22.73*/11.43*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.302s		22.93*/11.51*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.310s		23.10*/11.61*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.318s		23.25*/11.74*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.326s		23.39*/11.87*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.332s		23.52*/12.00*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.340s		23.64*/12.13*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.348s		23.73*/12.25*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.356s		23.78*/12.37*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.362s		23.81*/12.48*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.370s		23.80*/12.59*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.378s		23.75*/12.72*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.384s		23.68*/12.85*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.392s		23.57*/12.98*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.400s		23.44*/13.09*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.408s		23.27*/13.16*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.414s		23.05*/13.20*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.422s		22.78*/13.18*	tilt: 0.00/0.00	pressure: 0.00
 event30  TABLET_TOOL_AXIS        +1.430s		22.48*/13.13*	tilt: 0.00/0.00	pressure: 0.00
[... cut ...]
Group:            9
Seat:             seat0, default
Size:             54x14mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a```

`sudo libinput list-devices | grep -E 'Cap.*tablet'` returns:
```
Capabilities:     tablet
```

gonX, 2023-10-08:
Thanks for the follow-up, nothing from that output indicates what I suspected so this does need further investigating.

FWIW, Sway does not seem to exhibit any of these issues in Artist Mode, but I do also understand that Sway isn't exactly a drop-in replacement for neither Gnome nor KDE.