OpenTabletDriver/OpenTabletDriver #2935: No Pen Pressure in Absolute Mode

Capital-EX, 2023-09-28:
### Steps to Reproduce

Using Absolute Mode:
1. Set OTD to absolute mode
3. No pen pressure 

### Expected Behavior

I can access my tablet's pen pressure.

### Observed Behavior

I do not get pen pressure.

### Tablet Device

XP-Pen Artist 12 Pro

### Diagnostic Information

[debug.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12753285/debug.json.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-09-28:
Absolute mode is not designed to have pressure. This is not a bug, this is completely intended. Use artist mode instead.

Capital-EX, 2023-09-28:
I'm trying to use Absolute Mode because Artist Mode doesn't allow for mouse button bindings which I need for Krita such as opening it's right click menu. I didn't know if this was by design or not as the modes are not listed in the Wiki.

jamesbt365, 2023-09-28:
Then use artist mode mouse bindings (click the 3 dots to the right of a binding) or use key bindings instead.

Capital-EX, 2023-09-28:
The mouse bindings for artist mode don't work.

X9VoiD, 2023-09-28:
"Artist mode bindings" not "mouse bindings".

Capital-EX, 2023-09-28:
I have Artist Mode enabled. I went to Pen Settings > Pen Buttons > Those ... to the right of the binding.

It pulled up the advance binding editor. Using "Linux Artist Mode: (Button: Pen Button 1)" didn't have any affect. Nor did using "Mouse Button Binding: (Button: Middle)". 

I made sure to save and apply the setting changes each time to no results.

jamesbt365, 2023-09-28:
help > export diagnostics, save that and send it here.

flyingpie, 2024-09-09:
@jamesbt365 I'm running into the same issue.

Ultimately, I'd like to have pen pressure _and_ the ability to map pen buttons to mouse buttons. Is that not a thing by design, or am I running into an issue?

I've attached the diagnostics.

Thank you in advance!

[wacomdiag.json](https://github.com/user-attachments/files/16929678/wacomdiag.json)

jamesbt365, 2024-09-09:
It is a limitation by Linux itself, the only way to effectively get around this issue would be for OTD to "switch" the output mode, sync the cursor, then send the key press of your choosing, which would also prevent you from using the pen tip while this is happening.

Creating a workaround like this is just not in the scope at the moment.

flyingpie, 2024-09-09:
@jamesbt365 Thank you for your reply!

I can understand that you don't wanna go that path, I'll look for a different workaround until maybe a nicer solution pops up.

flyingpie, 2024-09-09:
@jamesbt365 One follow-up, would it be doable to have a key binding or something that toggles between artist- and absolute mode manually?

jamesbt365, 2024-09-09:
You do not need to mention me every single time you send a message, I get an email anyway and the webhook in discord makes it very clear that a message has been sent. Its just obnoxious.

Such bindings are available in `master` at the moment natively, but you will not get support for using dev builds based on master at this time.

You can use the `Preset Bindings` plugin inside the plugin manager to emulate this behaviour.

closed 2025-12-06 by gonX