OpenTabletDriver/OpenTabletDriver #2937: Support Request: Relative mode on Huion New 1060 Plus does not work

yiqubuhui, 2023-09-29:
### What do you need help with?

OTD's relative mode never worked for my Huion New 1060 Plus, the mouse mode of the factory driver works but the relative mode on OTD didn't, I much prefer the OTD for osu! and I can't really have two drivers at the same time, so I wanted to see if I could get the relative mode to work, thank! I linked a video of me using relative mode then VMulti relative mode(bad video quality sorry)

### What have you tried?

Before writing this ticket I didn't know that Huion 1060 plus and Huion new 1060 plus were different so I assumed my tablet was supported, then I found that the Huion new 1060 plus "Requires Zadig's WinUSB to be installed on interface 1", which I followed the instructions and setup but it still did not fix the problem of relative mode not working, I checked the FAQ the official wiki page, and I didn't find anything. I also installed the VMultiDriver and Vmulti Mode to see if it would help and it didn't either. Sorry in advance if I was just blind and didn't see a solution! :p 

### Diagnostic Information

[Diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12756126/Diagnostics.txt)

https://drive.google.com/file/d/19xz5nJ3yOukvvmyNxLkIMtegWumCW5u9/view?usp=sharing

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

gonX, 2023-10-01:
Hello, the linked video requires requesting access. I suggest uploading it to Youtube as an unlisted video instead.

Also, this smells a little bit like #2441 - but fortunately, it is usually suggested that you use absolute positioning for osu! gameplay.

yiqubuhui, 2023-10-02:
Hey! I made it a unlisted video, I was hoping to use my tablet to play some shooters is all, so I decided to try relative mode. The video is not too clear due to my phone camera, I was first on absolute mode and showing that it worked fine, then went to relative mode to show that the cursor was teleporting back to the starting position(not the same issue as in the FAQ) instead of moving from the current position, then I showed the same thing with VMulti relative mode, I'm happy to provide more details if needed(just incase you can't read the text because of the bad camera quality, also attached some images). Thanks!
[https://youtube.com/shorts/yu6JK0DcbC8?si=ISS0U4GD7yd1-g9G](url) 
[image]

[image]

[image]

jamesbt365, 2023-10-02:
You probably want to lower your reset time to a value lower if that is the behavior you are after.

yiqubuhui, 2023-10-02:
Just tried, that doesn't seem to be the problem, it's almost like even in relative mode, the tablet still functions as if its in absolute mode.

jamesbt365, 2023-10-02:
Export diagnostics again and send them here.

yiqubuhui, 2023-10-02:
[Diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12786765/Diagnostics.txt)

jamesbt365, 2023-10-02:
Could you:
- unplug the other tablet
- file > reset to defaults and save
- change back into relative mode and change your reset time to 15ms
- see if the issue still occurs.

yiqubuhui, 2023-10-02:
unplugged other tablet, reset to default, saved and applied, changed to 15ms, saved and applied, error still occurs, linked new diagnostic report if that helps

[Diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12786904/Diagnostics.txt)

[image]

gonX, 2023-10-09:
The video does still not work, BTW. It still requires me to request access.

Reopening issue since it does not seem solved

gonX, 2023-11-25:
A change has been submitted to fix relative mode. Please test with one of these builds:
https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/6882751398?pr=3005

yiqubuhui, 2023-11-25:
It is now working, tysm!

closed 2023-11-25 by yiqubuhui