OpenTabletDriver/OpenTabletDriver #2943: Better joystick detection avoidance for Linux Absolute Mode

gonX, 2023-09-30:
### Description

While #2939 does prevent us from being used as a joystick, I personally don't have a good feeling about this approach (removing the associated js endpoint). Instead, we should be adapting to `joydev.c`'s detection routines.

There are [fundamentally 2 checks](https://github.com/torvalds/linux/blob/9f3ebbef746f89f860a90ced99a359202ea86fde/drivers/input/joydev.c#L898) that makes joydev ignore a device:
1) It matches with their blacklist, either with:
    - [presence of `BTN_TOUCH` in `EV_KEY` typecodes](https://github.com/torvalds/linux/blob/9f3ebbef746f89f860a90ced99a359202ea86fde/drivers/input/joydev.c#L779-L784)
    - [presence of `BTN_DIGI` (aka `BTN_TOOL_PEN`) in `EV_KEY` typecodes](https://github.com/torvalds/linux/blob/9f3ebbef746f89f860a90ced99a359202ea86fde/drivers/input/joydev.c#L786-L791)
    - or that it matches VID+PID of some preexisting devices containing accelerometers (currently only console controllers) - which we won't be able to abuse
2) It matches their [absolute mouse detection](https://github.com/torvalds/linux/blob/9f3ebbef746f89f860a90ced99a359202ea86fde/drivers/input/joydev.c#L820).
    - The absolute mouse detection is a little more complicated. Right now, they:
        1) Check for a minimum amount of events (at least `EV_SYN`, `EV_KEY` and `EV_ABS`)
        2) Check specific axis count (`ABS_X` and `ABS_Y` **_only_**), 
        3) Check for specific buttons (`BTN_LEFT`, `BTN_RIGHT` and `BTN_MIDDLE` **_only_**)

In our current state, we do get matched on the amount of events and axises, but get discarded as we have too many buttons (SIDE and EXTRA, aka mouse4 and mouse5).

This means there are 4 ways to work around this, in rough order of realisticity:

1) Spoof that we're using `BTN_TOUCH`
        This does not seem to have any side effects in my limited testing

2) Spoof that we're using `BTN_TOOL_PEN`
        I did not test this, but I imagine this is more likely to get detected as a tablet, which is what we want to avoid with absolute mice (though the same argument could go for usage of `BTN_TOUCH`)

3) Remove the side buttons from Linux' Absolute Mode.
        This is the option with the least unknowns - we lose side buttons, but we also no longer get detected as a joystick.

4) Ask the joydev maintainer to also support usage of `BTN_SIDE` and `BTN_EXTRA`
        This is probably not ideal given that we likely want to expand button compatibility, though their detection routine could be changed to account for that.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

AkiSakurai, 2023-10-02:
1. Spoof that we're using BTN_TOUCH 
Clicks are not being registered in RetroArch with the udev input driver. Although this could potentially be considered a bug in RetroArch.
```
[WARN] [udev] Single-touch devices are currently untested!
[DEBUG] [udev] === UDEV_INPUT_DEVICE INFO DUMP ===
[DEBUG] [udev] 	ident = OpenTabletDriver Virtual Tablet
[DEBUG] [udev] 	devnode = /dev/input/event24
[DEBUG] [udev] 	type = ID_INPUT_MOUSE
[DEBUG] [udev] 	handle_cb = 0x558553f36ac0
```
2. Spoof that we're using BTN_TOOL_PEN
Systemd tags it with ID_INPUT_TABLET, and KDE recognizes it as a drawing tablet.

gonX, 2023-10-02:
Thanks for the testing. That leaves those 2 options as suboptimal for now.