OpenTabletDriver/OpenTabletDriver #2945: Add support for XP-Pen artist pro 16 2nd (not "artist pro 16", and not "artist 16 2nd")

m10d, 2023-09-30:
### Manufacturer

XP-Pen

### Product Name

MD160QH

### Device Width

1600px, 344.68mm (physical width 405.11)

### Device Height

2560px, 215.42mm, (physical height 291.37)

### Pressure Levels

16384 (I think this is a first?  I have no idea if it would ease getting support into a release, but I couldn't care less about 16k over 8k, especially not for a first release.)

### Lines per Inch (LPI)

5080 LPI

### Additional Information

https://www.xp-pen.com/product/artist-pro-16-gen-2.html

Product Name | Artist Pro 16 Drawing Display (Gen 2)
-- | --
Product Model | MD160QH
Color | Black & grey
Stylus | X3 Pro Smart Chip Stylus
Initial Activation Force | 3 g
Pressure Levels | 16384
Tilt | 60°
Display Resolution | 2560 X 1600
Dimensions | 405.11 X 291.37 X 20.23 mm
Work Area | 344.68 X 215.42 mm
Full Lamination | Yes
Viewing Angle | 178°
Contrast | 1200:1
Brightness (typical) | 250 cd/m2
Resolution | 5080 LPI
Color Gamut Coverage Ratio (typical) | 99% sRGB, 97% Adobe RGB, 99% DCI-P3
Color Gamut Area Ratio (typical) | 159% sRGB, 118% Adobe RGB, 117% DCI-P3
Reading Height | 10 mm
Response Rate | 20 ms
Report Rate | 200 RPS (max)
Accuracy | ±0.4 mm (center), ±0.8 mm (corner)
Port | 1 x Full-featured USB-C1 x 3-in-1 USB-C
Power Input | AC 110-240V
Power Output | DC 5V ⎓ 3A
Compatibility | Windows 7 (or later), macOS 10.10 (or later), Android (USB3.1 DP1.2), Chrome OS 88 (or later), Linux.

</div></div></div></div>

### Diagnostic Information

too long

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

m10d, 2023-09-30:
```
```
{
  "App Version": "OpenTabletDriver v0.6.3.0",
  "Build Date": "09/06/2023",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.15.0.84",
    "VersionString": "Unix 5.15.0.84"
  },
  "Environment Variables": {
    "USER": "xxxx",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/xxxx",
    "PATH": "xxxx"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.2/0003:046D:C52B.0003/hidraw/hidraw2",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50475,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.3/usb3/3-6/3-6:1.0/0003:048D:8297.0004/hidraw/hidraw3",
      "Manufacturer": "ITE Tech. Inc.",
      "ProductName": "ITE Device(8595)",
      "SerialNumber": "",
      "FriendlyName": "ITE Device(8595)",
      "VendorID": 1165,
      "ProductID": 33431,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0b:00.3/usb5/5-4/5-4.4/5-4.4:1.0/0003:045E:07A5.0005/hidraw/hidraw0",
      "Manufacturer": "Microsoft",
      "ProductName": "Microsoft® Nano Transceiver v2.1",
      "SerialNumber": "",
      "FriendlyName": "Microsoft® Nano Transceiver v2.1",
      "VendorID": 1118,
      "ProductID": 1957,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-3/1-3:1.2/0003:046D:C52B.0003/0003:046D:4082.0009/hidraw/hidraw6",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50475,
      "InputReportLength": -1,
[... 3977 lines cut ...]
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-09-30T17:04:12.6514273-06:00",
      "Group": "Settings",
      "Message": "Loaded preset 'L2Up2R: landscape-4k-32in under 1920x1080-tablet left of portrait-4k-32in.json'",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-09-30T17:04:12.6514329-06:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 1 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

m10d, 2023-10-28:
adding info from `discord`

- example of adding the config file: https://github.com/OpenTabletDriver/OpenTabletDriver/pull/2734/commits/84d55eaf1108f5f9b900c5f9cb6f1b1d29f46427
    - Match reportlengths and VID+PID. You may need temporarily add a udev rule as well so you can access your tablet from OTD
- To test locally, Put the file in either ~/.local/share/OpenTabletDriver/Configurations or ~/.config/OpenTabletDriver/Configurations - can't remember which on top of my head. You'll likely need to manually create the Configurations folder.
- add the aforementioned udev rule.

m10d, 2023-10-29:
this config "works"
- [0,0] - [~32k,~32k] range
- 0~16k pressure
- correct T/F on the 2 pen buttons and the eraser
- correct tilt

range is [0,0]~[32k,32k]

in the config UI, snapping the `Dispaly` to the XPPen, and leaving the `Tablet` square as it is
- the pen upper-left matches the pointer
- but dragging diagonal across the tablet:
    - drag pen 100% to the right, cursor only moves 50% of way across tablet
    - drag pen 100% to bottom, the cursor only translates ~80% of the way down/Y the tablet

```
{
  "Name": "XP-Pen Artist 16 Pro (2nd Gen)",
  "Specifications": {
    "Digitizer": {
      "Width": 344.68,
      "Height": 215.42,
      "MaxX": 68936.0,
      "MaxY": 43084.0
    },
    "Pen": {
      "MaxPressure": 16384,
      "ButtonCount": 2
    },
    "AuxiliaryButtons": null,
    "MouseButtons": null,
    "Touch": null
  },
  "DigitizerIdentifiers": [
    {
      "VendorID": 10429,
      "ProductID": 2395,
      "InputReportLength": 10,
      "OutputReportLength": null,
	"ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser",
      "FeatureInitReport": null,
      "OutputInitReport": [
        "ArAE"
      ],
      "DeviceStrings": {},
      "InitializationStrings": []
    }
  ],
  "AuxiliaryDeviceIdentifiers": [],
  "Attributes": {
    "libinputoverride": "1"
  }
}
```

here's my layout (on X11 / xrandr)
```
#      (== 3840
#         -1600
#         -2160)
#  zero  1280       3840
#   |     |          |
#   ;     V          V
#   ;                            .
#   ;                |--------| <- zero
#   ;     |----------|        | <- 80
#   V     | 2560x1600|  SMSNG |
#         |  xppen   |        |
#   |----------------|  2160  | <- 1680 (==3840-2160)
#   |      ACER      |  -3840 |
#   |    3840x2160  ||- - - - | <- 2240 (==3840-1600)
#   |               ||        |
#   |---------------||--------| <- 3840
```

[image]

gonX, 2023-10-30:
> note I do not understand the values in the UI - after set to display, Display:X,Y is 2560,880 ... as above with these values, the upper left pen on tablet matches the cursor; and if I set it to the [1280,80] I calculate the pen/cursor never match, but what do these mean?

Offsets are calculated by center origin, ie. X = 2560 makes sense because your 2560 is right-aligned to the 3840 pixel origin, ie. your target monitor starts `3840 - 2560 = 1280` pixels in. Because the size of the display is 2560, you add half of that to the offset to get the center, ie. 1280+(2560 / 2) = 2560, so X matches.

With Y it also matches because your target monitor starts 80 pixels down, and the center height is `1600 / 2 = 800`, ie. Y = 880.

As for the scale and max coordinates not working, this is caused by the parser being incorrect. Of the final parsers for XP-Pen we have that supports higher than 8 bits of scale (ie. `signed short`/`sbyte`), only `XP_PenOffsetPressureReportParser` works, however, you should also verify that the pressure reported with the new parser is correct.

Therefore, if you do the following change, the full area of the tablet _should_ work:
```diff
- "ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser",
+ "ReportParser": "OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenOffsetPressureReportParser",
```

m10d, 2023-10-30:
> As for the scale and max coordinates not working, this is caused by the parser being incorrect. Of the final parsers for XP-Pen we have that supports higher than 8 bits of scale (ie. `signed [...]

switching to `XP_PenOffsetPressureReportParser` does not appear to help the range issue (It still seems to max at 32k, and the cursor behavior after setting the UI is the same as above: 100% pen movement translates to x~=50%, y=~80% cursor movement.

While I don't see it in the logs to confirm - I can be pretty confident it picked up my changes as the pressure is broken with a signed-int issue it appears: pen pressure reads ~4billion ~ 8000.

Side-note: as soon as I actually use the pen on the tablet in this config, my mouse click becomes broken.  I can still move the mouse but can't click on anything.  My mouse cursor sometimes gets stuck as the pen cursor, too.  Normal behavior is restored when I use the keyboard (woot for `i3` :) ) to open a terminal and `systemctl stop` the the service.

m10d, 2024-07-18:
@gonX - I was hoping I could ask for your thoughts on if there are any behind-the-scenes changes that may have addressed the issues above - or any thoughts on if/when this might be addressable.

To summarize the history (not being versed in the code at all, please forgive if this isn't correct) it seems from my previous tests that the 16k pen itself is parsed correctly _in some paths_; but the `XP_PenOffsetPressureReportParser` regressed that (pressure broken switching to it from `XP_PenReportParser`, but X/Y scaling issue wasn't fixed or affected.

I'm more than happy to test something new if you believe there might be a change.

Thanks for your help & the project.