OpenTabletDriver/OpenTabletDriver #2946: Click Doesnt register

osuaimer, 2023-10-01:
### Steps to Reproduce

Like the title says clicks dont register, i have tried in absolute mode, relative ,vmulti relative etc nothing but when i go to tablet debugger there is pressure registering 
[image]

[image]
i have tried playing around with tip thresh hold using every number on the slider still nothing

### Expected Behavior

When i press down on my tablet it should left click

### Observed Behavior

nothing happens 

### Tablet Device

Wacom 672

### Diagnostic Information

System.UriFormatException: Invalid URI: The URI is empty.
   at System.Uri.CreateThis(String uri, Boolean dontEscape, UriKind uriKind, UriCreationOptions& creationOptions)
   at System.Uri..ctor(String uriString)
   at OpenTabletDriver.UX.MainForm.ExportDiagnostics() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\MainForm.cs:line 641

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

gonX, 2023-10-01:
Just to make sure, you're unable to click while using VMulti? If so, you need to change the click bindings to use VMultiMode. See the wiki page on VMultiMode here: https://github.com/X9VoiD/VoiDPlugins/wiki/VMultiMode

osuaimer, 2023-10-01:
ahhh thank you fixed it

closed 2023-10-01 by gonX