OpenTabletDriver/OpenTabletDriver #2949: Add support for

asluizh, 2023-10-02:
### Manufacturer

Wacom

### Product Name

HID 52FA

### Device Width

286

### Device Height

179

### Additional Information

This is the device integrated into the Lenovo Yoga 6. Here's the output of libinput list-devices:
```
Device:           Wacom HID 52FA Pen
Kernel:           /dev/input/event5
Group:            6
Seat:             seat0, default
Size:             286x179mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a

Device:           Wacom HID 52FA Finger
Kernel:           /dev/input/event6
Group:            6
Seat:             seat0, default
Size:             286x179mm
Capabilities:     touch 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      identity matrix
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   n/a
Rotation:         0.0
```

### Diagnostic Information

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12778857/diag.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-02:
If I remember correctly somebody tried parsing the data from one of these a while ago, but has yet to open a pull request. Until a pull request has been opened I am not personally putting any effort to create a configuration for this tablet.

gonX, 2023-10-08:
The linked diagnostics does not have any HID devices in it. According to [the PR adding support for it in libwacom](https://github.com/linuxwacom/libwacom/pull/536/files#diff-e95fd7f03777d90fd1417e25304363c3c671c8b296e43df4204489073222f986R12), the pen is I2C and not USB, which means that this relies on #1205

jamesbt365, 2025-01-24:
Closing until a backend for OTD to support this is added, which is currently out of scope but PRs are welcome. Issue tracking I2C support is #1205.

closed 2025-01-24 by jamesbt365