OpenTabletDriver/OpenTabletDriver #2955: Provide an arm or universal build for macOS.

foldfree, 2023-10-05:
### Description

It would be great to have a native arm aarch64 build for apple silicon devices running on macOS.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-05:
The x86 builds work perfectly fine with the compatibility layer in MacOS, so this isnt a high priority at the moment.

From previous testing and if I remember correctly these builds needs to be signed on MacOS which would mess with our CI. Somebody correct me on that if I'm wrong.

foldfree, 2023-10-06:
Not sure if this is what you mean but it is still possible to run unsigned builds on arm macOS by using `xattr -dr com.apple.quarantine /path/to/program.app`

jamesbt365, 2023-10-06:
And that is extremely inconvenient for users, far less than just packaging x86 builds and only having one binary for all of MacOS.

scj643, 2024-12-23:
Building from source using the ARM .net works just fine. Also if needed I could codesign it.

jamesbt365, 2024-12-23:
ARM builds are impossible until 0.7.0.0 due to a problem with our UX, afaik `0.6.x` (the branch all releases are currently built upon) doesn't work.

scj643, 2024-12-23:
```
Unhandled exception. System.EntryPointNotFoundException: Unable to find an entry point named 'objc_msgSend_stret' in shared library '/usr/lib/libobjc.dylib'.
   at MonoMac.ObjCRuntime.Messaging.CGRect_objc_msgSend_stret(CGRect& retval, IntPtr receiver, IntPtr selector)
   at MonoMac.AppKit.NSScreen.get_Frame()
   at Eto.Mac.Forms.MouseHandler.get_Position()
   at Eto.Forms.Mouse.get_Position()
   at OpenTabletDriver.UX.MainForm.InitializeForm() in /Users/scj643/devel/OpenTabletDriver/OpenTabletDriver.UX/MainForm.cs:line 87
   at OpenTabletDriver.UX.MainForm..ctor() in /Users/scj643/devel/OpenTabletDriver/OpenTabletDriver.UX/MainForm.cs:line 32
   at OpenTabletDriver.UX.App.RunInternal(String platform, CommandLineOptions options) in /Users/scj643/devel/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 55
   at OpenTabletDriver.UX.App.Run(String platform, String[] args) in /Users/scj643/devel/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 41
   at OpenTabletDriver.UX.MacOS.Program.Main(String[] args) in /Users/scj643/devel/OpenTabletDriver/OpenTabletDriver.UX.MacOS/Program.cs:line 13
```

Yep the UI just dies :(

Still would be possible to have the daemon be a universal binary though

joaofrgomes, 2025-09-13:
Hi, I'm the user who just mentioned this issue. Any updates on this front?

I don't have any numbers from proper marketing research on me, but considering how big the performance gap between Intel and Apple Silicon machines was from the get-go and how even bigger it has become since the first M1 machines dropped, pure anecdotal evidence from seeing all my designer buddies canning the latter and adopting the former, and [the announced demise of OCLP](https://github.com/dortania/OpenCore-Legacy-Patcher/issues/1167), which will further force the big old hardware stalwarts (myself included, I'll have to upgrade my daily drivers, demote my oldest Apple Silicon machines to emergency backups, and move all my old Macs, which were indeed my backups, to x86-64 Linux distros, to keep them relevant and useful or something) to start considering upgrades too… there's a perfect storm forming.

Older, unmaintained x86-64 macOS builds and [the imminent demise of Rosetta 2](https://developer.apple.com/documentation/apple-silicon/about-the-rosetta-translation-environment/) will mean that in one, maybe two years at best, x86-64 macOS-only support in OTD will become untenable and make it too much of a niche solution for Mac users, instead of making it **the universal, go-to one**, as Wacom smells the blood on the water from their own forced obsolescence and reaps the benefits from users who'll prioritize having an updated, secure OS on their daily driver over saving on buying a new tablet and inevitably cave in.

Please bear in mind, and I cannot stress this enough, that I'm not hostile or indifferent to your efforts, or to the mindset behind them; I love FOSS solutions and have been using them as much as I possibly can, also contributing with my own beta testing user feedback (Zotero and ODF's LibreOffice immediately come to mind), just asking the hard questions and stating the hard truths here. In that vein, I also would very much like to help you, but preferably on a solution that is guaranteed to work long-term, with the least hiccups possible, and benefit, well, me as well, that being **native Apple Silicon builds of OTD**. I know it's terrible to have to invest on new hardware, address the whole “UX breaking” conundrum (excuse me my ignorance; I'm not too comfortable with having to build stuff from source, so I haven't gone that route just yet), the works, but such is the way of the world…

FWIW, I own a Wacom CTH-670, and will test it today with OTD on the Tahoe RC build I've been running recently (@thenickdude's fixed drivers seem to work fine, and will likely be my preferred solution until they inevitably break, to be honest, but I can see the writing on the wall and welcome my new FOSS overlords).

joaofrgomes, 2026-09-02:
Update: Apple is now getting more aggressive in finishing their transition, by allowing (and strongly encouraging) developers to drop Universal app submissions on their Mac App Store and going for Apple Silicon-only binaries, and reminding everyone that “If you haven’t updated your app yet: **Begin the transition immediately**. Build and distribute your app as a universal binary, which runs natively on both Apple silicon and Intel-based Mac computers.” on a [new support document](https://developer.apple.com/news/?id=w5ngl9k2).

We still don't know what vestiges of Rosetta 2 will remain or what kind binaries they'll support, but the same document also states that macOS 27 is the “Final release to support Rosetta — Intel-only apps will no longer run on Mac computers with Apple silicon after this update.”.

If anything, this should change the priority on this from “low” to something else a bit higher and prompt at least one of the devs to snatch an old M1 machine (yes, _old_; the first models are three months away from turning six years old at this point, and about to become vintage even by AppleCare standards) from a thrift store.

I am more than happy to help as an end user with access to an M1 Max Mac Studio and an M2 MacBook Air, but at some point, if you want to create and maintain software that people can actually use indefinitely, you **have** to stay current, because peripherals last and stay useful way longer than desktops and laptops and even people who wish to save the former from the e-waste bin they may be forced, for competitive or even security reasons, to upgrade their daily drivers.

Kuuuube, 2026-09-14:
Status update on this for anyone curious.

With https://github.com/OpenTabletDriver/OpenTabletDriver/pull/4943 being merged as well as https://github.com/OpenTabletDriver/OpenTabletDriver/pull/4721 and https://github.com/OpenTabletDriver/OpenTabletDriver/pull/4730 for signing support, we currently do have support for ARM/Apple Silicon macOS builds. It isn't piped into the CI or releases at the moment but it is available and working in the build system. There is no need to worry about not being able to use OTD after Apple decides to drop emulation support. We will have builds ready in time for the date they've set but even if they jump their deadline, we are set up to provide dev builds or allow users to easily build it themselves.

Apple has stated that intel app emulation will be [discontinued in macOS 28](https://support.apple.com/en-us/102527) and based on prior release history this version should be expected to release around September 2027.

alf239, 2026-10-17:
Data point and a test offer, since the last status said the blocker is nobody on the team having an Apple Silicon machine.

**Tested today:** v0.6.7 `osx-x64` on an M-series Mac running macOS 27.0.1 with a Wacom Intuos 3 (PTZ-630, `056a:00b1`). The daemon starts fine under Rosetta, but ends with `No tablets were detected.` with no error and no permission prompt. The cause is that the process never gets **Input Monitoring**: `IOHIDManagerOpen` on a mouse-class HID device returns `kIOReturnNotPermitted` (`0xe00002e2`) until the user grants it, and nothing in the daemon calls `IOHIDRequestAccess(kIOHIDRequestTypeListenEvent)`, so macOS never shows the dialog. A one-line call there (or logging the `NotPermitted` result) would turn a silent failure into an actionable one, independent of the arm64 question.

**Offer:** I'm happy to test `osx-arm64` builds here (macOS 27, Apple Silicon, Intuos 3 for pressure/tilt/eraser checks) and report back. `build-matrix.yml` on `0.6.x` already has the signed `osx-arm64` entry, so if a CI artifact or dev build is available, point me at it.

Mrcubix, 2026-10-17:
It is indicated in the install doc for MacOS : https://opentabletdriver.net/Wiki/Documentation/RequiredPermissions#macos

The permission request also exist, it is the first call before showing the UX : 

https://github.com/OpenTabletDriver/OpenTabletDriver/blob/e8b6faa23b37df872be7f9b9c2e47d2f87368235/OpenTabletDriver.UX.MacOS/Program.cs#L10
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/e8b6faa23b37df872be7f9b9c2e47d2f87368235/OpenTabletDriver.UX.MacOS/PermissionHelper.cs#L27-L46

alf239, 2026-10-17:
Thanks, and apologies: you're right on all three. I had started the daemon on its own, which skips `PermissionHelper`, and I missed the macOS section of the permissions page.

Test report for `OpenTabletDriver-0.6.8-rc1_osx-arm64` on an M-series Mac, macOS 27.0.1, Wacom Intuos 3 PTZ-630 (`056a:00b1`):

- Native arm64 daemon and UX start, the UX requests both permissions and relaunches, the tablet is detected ("Found tablet 'Wacom PTZ-630'") and the default Absolute Mode profile works.
- Krita 5.3 Tablet Tester: `Pen tip brought near`, `Stylus press/move` with pressure (5 to 75% on a test stroke), tilt, eraser recognised. Tayasui Sketches: pressure-sensitive strokes. A global `NSEvent` monitor confirms the TabletPoint subtype with pressure and tilt, and proximity events for pen (`vendorPointerType 0x802`) and eraser.
- Nothing to report against the build. Happy to test further rc builds on this hardware.

closed 2026-10-17 by Kuuuube