OpenTabletDriver/OpenTabletDriver #2956: Support Request: intuos M problem

SkyMoon4106, 2023-10-07:
### What do you need help with?

Hi everyone, I am a Chinese speaker. Please forgive me for using very bad English.
I connected my intuos M (ctl 6100)
Also installed the wacom driver
But opentabletdriver did not detect it

screenshot&photo
[https://i.imgur.com/U08knfY.jpg](url)
[https://i.imgur.com/9TXWuXV.png](url)
[https://i.imgur.com/jfF0kQV.png](url)

### What have you tried?

I've tried everything on the wiki, but nothing works

### Diagnostic Information

N/A

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2023-10-07:
> Also installed the wacom driver
> But opentabletdriver did not detect it

OTD is not detecting your tablet because you installed the wacom driver. You can remove everything all in one go with [this](https://github.com/X9VoiD/TabletDriverCleanup/releases/latest).

Next time you open an issue, diagnostic information *does* apply, so follow the instructions and provide the information we request. Thank you.

closed 2023-10-07 by jamesbt365

SkyMoon4106, 2023-10-07:
I tried deleting the wacom driver and starting the OTD again, but it didn't work either.

SkyMoon4106, 2023-10-07:
Sorry, after rebooting he detected