OpenTabletDriver/OpenTabletDriver #2960: Wacom Intuos CTH-690: Straight lines going backwards at the ends of strokes

lily-00, 2023-10-15:
### Steps to Reproduce

1. Install OpenTabletDriver using the official instructions on any OS, for example Arch Plasma Wayland
2. Open any drawing application, for example Krita
3. Using the Wacom Intuos, make any brush stroke on the canvas

### Expected Behavior

The drawn brush stroke should reflect the movements that were done with the pen

### Observed Behavior

The end of every stroke is a straight line going backwards to where the pen used to be a moment ago, so parallel to the stroke if I draw a straight line and at bigger angles if I draw a curved line.

[image]

[image]

I use Arch Plasma Wayland but I tried it on X11 and Wayland, Plasma and GNOME, Arch and Ubuntu 22.04.3, as well as Windows 11, using Krita, Aseprite, Blender and the GNU Image Manipulation Program. It happens everywhere.

The same issue also happens with xf86-input-wacom if the “raw sample rate” slider in the Plasma tablet settings is set to 1 instead of the default 4, so both drivers may have the same issue (see https://github.com/linuxwacom/xf86-input-wacom/issues/319). It does not happen on Wayland with libinput or on Windows 11 with the official Wacom driver.

See also my thread at https://krita-artists.org/t/wacom-intuos-analysis-of-straight-lines-at-the-beginnings-or-ends-of-strokes/75702

### Tablet Device

Wacom Intuos CTH-690

### Diagnostic Information

[opentabletdriver-diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12909199/opentabletdriver-diagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

hwsmm, 2023-10-15:
This is already a known issue in the OTD community. It's because x90 tablets only apply smoothing when the pen is hovering, and it causes weird hover/drag transition you are having now.

This is a hardware issue, not a driver issue. You can mitigate this by using some smoothing plugins which you can find from plugin manager.

jamesbt365, 2023-10-15:
As hwsmm has said, this is a bug in the firmware of the tablet, not a problem with OTD, but doesn't mean it can't be mitigated in software. You can follow the advice in #2806 (your issue is a duplicate of this)

closed 2023-10-15 by jamesbt365