OpenTabletDriver/OpenTabletDriver #2969: vk640 graphics  tablet button not working (pen button working), and pen presser dont work

View-my-Git-Lab-krafi, 2023-10-20:
### Description

vk640 graphics  tablet button not working,
actually i am not sure is it a bug or feature not included there is only pen setting option 
and options are 
[image]

 pen presser dont work 
i have tried a lot of plug in and expecting maybe it will fix my issue but didnt work at all 

[image]

i know its disable bcz it dont work so i put it disable 

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

View-my-Git-Lab-krafi, 2023-10-20:
even though if developer dont have plan to include those feature just tell me how can try to do that  by myself give me some other graphics tablet programming example

jamesbt365, 2023-10-20:
OTD version?

jamesbt365, 2023-10-20:
Looking back on the code this was added over a year ago at this point. Update to the latest version of OTD. This was fixed way back in #2294.

closed 2023-10-20 by jamesbt365